dhrone/pydPiper #70: Errors with squeezelite and lms server on a different host

paolobormida, 2019-09-18:
Hi! I have a RASPDAC installed with the image available on the Audiophonics blog page here

https://www.audiophonics.fr/en/blog-diy-audio/23-start-up-raspdac-rasptouch-raspdac-mini-with-a-pre-configured-image.html

I am using the RASPDAC DietPi image, and I have disabled LMS since I have another LMS running on the network. I have also modified /etc/systemd/system/squeezelite.service adding -s option to connect it to the LMS server running on my network.
Then I have modified /lib/systemd/system/pydpiper.service to specify the MAC address of the local interface squeezelite is using to connect to LMS server.

When pydpiper.service starts, it records this message in the log (showing output of systemctl status here):

root@DietPi:~# systemctl status pydpiper.service 
● pydpiper.service - pydPiper
   Loaded: loaded (/lib/systemd/system/pydpiper.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-09-18 16:43:13 BST; 9min ago
 Main PID: 1212 (docker)
      CPU: 298ms
   CGroup: /system.slice/pydpiper.service
           └─1212 /usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/dietpi/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py --lms --lmsplayer b8:2
7:eb:f2:71:b3 --driver winstar_weg --width 80 --height 16 --rs 7 --e 8 --d4 25 --d5 24 --d6 23 --d7 27 --timezone Europe/Paris --temperature celcius --pages pages_raspdac_16x2_lms.py

Sep 18 16:43:34 DietPi docker[1212]: Traceback (most recent call last):
Sep 18 16:43:34 DietPi docker[1212]:   File "/usr/lib/python2.7/threading.py", line 552, in __bootstrap_inner
Sep 18 16:43:34 DietPi docker[1212]:     self.run()
Sep 18 16:43:34 DietPi docker[1212]:   File "/usr/lib/python2.7/threading.py", line 505, in run
Sep 18 16:43:34 DietPi docker[1212]:     self.__target(*self.__args, **self.__kwargs)
Sep 18 16:43:34 DietPi docker[1212]:   File "/app/sources/musicdata_lms.py", line 136, in run
Sep 18 16:43:34 DietPi docker[1212]:     self.connect()
Sep 18 16:43:34 DietPi docker[1212]:   File "/app/sources/musicdata_lms.py", line 108, in connect
Sep 18 16:43:34 DietPi docker[1212]:     raise RuntimeError(u"Could not find any LMS Players")
Sep 18 16:43:34 DietPi docker[1212]: RuntimeError: Could not find any LMS Players

Is it supported to run squeezelite and pydpiper on an host and lms on another host and have the display update accordingly while music plays?

Thanx, Paolo

dhrone, 2019-09-18:
It is absolutely possible to have a separate LMS server that you can pull the metadata from.  You should verify that you only have one LMS server on the network.  When multiple LMS servers exist on the same network segment, the LMS players do not predictably associate with the correct server.

FYI, typically the MAC address of the LMS player is not specified in the service file.  Normally this would be configured in pydPiper.cfg which is created by the configure.py program.  You will need to make sure that you are also specifying the correct IP address (or URL if DNS is configured) for the LMS server that you are using within the pydPiper.cfg file.  Again, this is set when you run configure.py.

paolobormida, 2019-09-18:
This are the only instructions that comes with the RASPDAC image from Audiophonics:

```
CUSTOM RASPDAC DIETPI IMAGE

- I-Sabre ES9028/38 Driver with hardware volume
- Pydpiper OLED SPI driver for LMS
- LIRC Remote script
- LMS pre-installed with Qobuz plugin, and hardware volume setting.
- Samba pre-installed 

Pydpiper require Squeezelite MAC adress setting in : /lib/systemd/system/pydpiper.service
LIRC require Squeezelite MAC adress setting in : /etc/lirc/lmscli.exp
Samba require NAS configuration with : "dietpi-drive_manager"

-- Credits --
Dietpi : http://www.dietpi.com
Driver I-sabre 9028/38 : https://github.com/audiophonics/Volumio2_I-sabre9028_plugin
Pydpiper : https://github.com/dhrone/pydPiper
```

So probably that is all that is needed when LMS is on the same host and the players, and it does not mention configure.py. Where can I find configure.py and run it?

Thanx, Paolo.

dhrone, 2019-09-18:
Hmm.  I suspect from your question that you are not running the latest commit of the code.  Can you let me know what version you are on?  

You may want to just start a brand new installation using the latest copy of the code.

paolobormida, 2019-09-18:
OK, I managed to change the pydPiper.cfg directly and changed locahost to my server fqdn per lms_server entry and lms_player to the mac address of the local player running squeezelite.
Everything is working fine now.

paolobormida, 2019-09-18:
OK, updating to the last versione would be interesting, how can i check what version is running on the RASPDAC image?

README.md that comes along with the pydPiper.py script say:

## History

Version 0.3 (Alpha). Moved pydPiper into a docker container.  Added support for HD44780 I2C displays and SSD1306 displays.
Version 0.251 (Alpha). Fixed Unicode bug in LMS driver.
Version 0.25 (Alpha). Added support for HD44780 style LCD displays.
Version 0.21 (Alpha). Minor bug fixes.
Version 0.2 (Alpha).  Initial testing release.

dhrone, 2019-09-19:
Unfortunately, my release management discipline has not been high and I am not actively maintaining the code at the moment.  I may get back to it later in the year though.

The easiest way to get to the latest code would be to create a new installation (fresh OS install) and then run through the installation script appropriate for the distribution you use.  It's largely...

Get pipenv up and running
Use pipenv to install the packages that pydPiper relies upon
```
pipenv install moment python-mpd2 pyLMS redis pyOWM luma.oled socketIO-client Pillow smbus
```
Clone the current master branch
```
git clone https://github.com/dhrone/pydPiper
cd pydPiper
```

Run the configuration script to create the pydPiper.cfg file.
```
pipenv run python2 configure.py

```

And then run pydPiper
```
pipenv run python2 pydPiper.py
```
If it runs correctly, you can then add it to systemv to get it to autostart.

GG700, 2022-05-14:
Hi,

I have a problem with a squeezelite and lms server on a different host.

My status : 
<pre>root@DietPi:/home/dietpi/pydPiper# systemctl status pydpiper.service
<font color="#8AE234"><b>●</b></font> pydpiper.service - pydPiper
   Loaded: loaded (/lib/systemd/system/pydpiper.service; enabled; vendor preset: enabled)
   Active: <font color="#8AE234"><b>active (running)</b></font> since Sat 2022-05-14 08:34:50 CEST; 48s ago
 Main PID: 13864 (docker)
    Tasks: 11 (limit: 4915)
   Memory: 20.0M
      CPU: 199ms
   CGroup: /system.slice/pydpiper.service
           └─13864 /usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/dietpi/pydPiper:/app:rw dhrone/pydpiper:v0.31
-alpha python /app/pydPiper.py --lms --lmsplayer B8:27:EB:34:F7:B0 --driver winstar_weg --width 80 --height 16 --rs 7 --e 8 --d4 25 --d5 24 --d6 
23 --d7 27 --timezone Europe/Paris --temperature celcius --pages pages_raspdac_16x2_lms.py

mai 14 08:34:50 DietPi systemd[1]: Started pydPiper.
mai 14 08:34:58 DietPi docker[13864]: pydPiper starting
</pre>

My config in the file pydpiper.cfg : 
<pre><span style="background-color:#FFFFFF"><font color="#3F3F3F">[SOURCE]</font></span>
<span style="background-color:#FFFFFF"><font color="#3F3F3F">source_type = lms</font></span>
<span style="background-color:#FFFFFF"><font color="#3F3F3F">lms_server = 192.168.0.6</font></span>
<span style="background-color:#FFFFFF"><font color="#3F3F3F">lms_port = 9090</font></span>
<span style="background-color:#FFFFFF"><font color="#3F3F3F">lms_user =</font></span>
<span style="background-color:#FFFFFF"><font color="#3F3F3F">lms_password =</font></span>
<span style="background-color:#FFFFFF"><font color="#3F3F3F">lms_player = B8:27:EB:34:F7:B0</font></span>
</pre>

Can you help me ?