dhrone/pydPiper #71: Problem with WINSTAR OLED WEH002004AR (20x4, RED)

mariuszmysz, 2019-09-20:
Hi!
I have a problem with the WINSTAR OLED 20x4 WEH display configuration. I have tested five configurations:

hd44780 + pages_lcd_20x4 (pulseEnable = 200, Animation = 0.15):

- the first initialization is correct; restart of the service causes that often the display remains in such a state that an attempt to reinitialize ends with the display of random characters; only disconnecting and connecting the power supply removes the problem; not suitable for mounting in a housing, unfortunately;

hd44780_i2c + pages_lcd_20x4 (pulseEnable = 200, Animation = 0.15):
- like hd44780, except that the frequency of incorrect initializations is low; however, it strongly depends on the current state of the display - if it is just scrolling - it is 99% that re-initialization will fail;

winstar_weg + pages_weh_80x16 (pulseEnable = 200, Animation = 0.15):
- perfect operation, 100% correct initialization after service restart - however, only 80x16 characters used - two display lines remain unused;

winstar_weg + pages_lcd_20x4 (pulseEnable = 0.1, Animation: from 0.0 to 1s, 100x32):
- problem with the "Animation Smoothing" parameter - unbearable blinking of displayed characters; reducing the parameter value to 0 slightly improves the situation, but still the image vibrates in a noticeably significant way;
- no response to the 100x32 setting - only the first two display lines are still used;
- setting effect pages_20x4 - time with BIG font is displayed in two lines - in an interesting way: top half - in the second line, and the bottom half in the first line;

winstar_weg + pages_lcd_16x2 (pulseEnable = 0.1, Animation: from 0.0 to 1s)
- for 80x16 setting - excellent, but in two lines;
- for 100x32 setting - display flickering, only two lines used, but correct interpretation of large fonts;

It is clear to me that the malfunction in hd44780 is related to a different LCD initialization sequence than for OLED.

**How can I force the winstar_weg driver to interpret the 20x4 (100x32) setting correctly?**
**Can I transfer a fragment of the display initialization code from winstar_weg to hd44780 "just like that"?**

In general, a great idea for software. I am so impressed that I prepared the housing (laser + CNC). The remaining OLED issue. Of course, I can buy an LCD display, but OLED looks cool and more suited to such a device as a network player.

dhrone, 2020-02-15:
You need to use the winstar_weg driver to get correct operation of your display.  Your problem is likely the pages file, not the display driver. 

I would recommend using the pages_lcd_20x4 file but with a small modification.  You may be able to get rid of the blinking by changing the scroll settings for the widgets that are set to scroll.  The pages file example for LCDs shifts each character by 5 pixels for every scroll action to keep the characters aligned with the underlying character structure of the HD44780 display.  This is not necessary for fully graphically displays like the winstar weg series.  If you take a look at the winstar_weg example page file you see that the scroll is only one pixel at a time which will appear much smoother.  So, use pages_lcd_20x4 but anything that says...

`'effect':('scroll','left',5,1,20,`

into...

`'effect':('scroll','left',1,1,20,`

mariuszmysz, 2020-02-25:
Hi dhrone,
Thank you for your response. I didn't know if you would answer. And, looking forward to answering, I forced the hd44780_i2c driver to work. This required small code changes in the initialization sequence, but now works without problems.
I have another question, but not related to the display anymore so I will open a new issue.

dhrone, 2020-03-03:
Would it be possible for you to share the changes you made to the driver?

mariuszmysz, 2020-04-05:
Hi, dhrone
of course I will share the code. I changed:

- display initialization section in the hd44780_2c class constructor (of course it doesn't look as aesthetically as you did, but I think it is readable)
- lcd_toggle_enable() procedure - instead of delayMicroseconds() I used time.sleep()
- clear() procedure - as above - time.sleep()

I did tests on rpi 2b + / 3b +. Thanks to this I removed such problems as: "garbage" on the display, incorrect position of the character on the display (eg: strings of characters were displayed from 3 columns), incorrect subsequent initializations of the display, which could be removed only by disconnecting the display's power supply (there is an example in the network of how to control the power supply of the OLED display by the MOSFET transistor, but I thought it was a desperate step - it just needs to work without the transistor).

Attached is the hd44780_i2c.py file with changes:
[hd44780_i2c.txt](https://github.com/dhrone/pydPiper/files/4434333/hd44780_i2c.txt)
I have not tested changes with LCD.

Of course, I didn't think of it myself. I used the code presented here:

[https://www.raspberrypi.org/forums/viewtopic.php?t=68055](url)

by user robertcx.

In addition, I attach a video where I present the display after changes:
[https://drive.google.com/file/d/1FY8d5IyLpejNOFNqgl7b9SKnRDZx2c02/view?usp=sharing](url)

Karsten66, 2020-11-18:
Hello,
I have the following problem. Volumio runs on my Raspberry Pi 4 / Raspberry Pi 3 with a 20x4 I2C display. Everything is okay and looks really great - thank you very much to you, you are doing a very good job.
Now I would like to get a Winstar OLED 20x4 WEH002004A (6800, 8080, SPI) display with pydPiper to work. Is it enough to solder an I2C module to the board of the WEH002004A, or do I have to connect to the Raspberry Pi via serial? How can I control the OLED? Can you please help?
Greeting,
Karsten

dhrone, 2020-11-18:
I wouldn't bother with the i2c module.  Just wire it up as a 4-bit parallel interface (e.g. 6800).  However, per this issue, you will need a modified version of the hd44780.py module.  Let me look into pulling that together for you.

Also, apologizes to @mariuszmysz.  I'll also include putting in a version of his fix for people using the i2c style interface with the 4 line WEH style displays.  FYI, my advice to him to use the winstar weg interface would not have worked as I've since learned that the 4 line WEH displays cannot be driven graphically like the 2 line versions can.

Karsten66, 2020-11-19:
Hello Dhrone,
many, many thanks for your quick reply. 
Since I'm a newcomer, I would like you to send me a circuit diagram for my Winstar with a simple 4-bit parallel connection to the Raspberry Pi if you have one. I spent half the night finding such a circuit on the Internet, nothing.
And can you also send me a modified version of the hd44780.py module?
It would be great if you could help me.
Thank you very much for your help.
Greeting,
Karsten

dhrone, 2020-11-20:
You can wire according to the instructions at https://luma-oled.readthedocs.io/en/latest/hardware.html#parallel.  Note, if you are using (or have enabled) SPI on your PI, you cannot use GPIO 7 and 8.  If you are, I'd use use GPIO17 (P01-11) for RS and GPIO 22 (P01-15) for E.  

I've just pushed a branch to pydPiper called weh002004.  If you pull it down, you can re-run configure.py.  The new driver name is winstar_weh.  You should make sure to choose a page file that is appropriate for a character-mode device (e.g. pages_lcd_16x2.py, pages_lcd_20x4.py).

Karsten66, 2020-11-21:
Hello,
thank you so much for your help.
Yes you are right, I use SPI on my Raspberry Pi. I used GPIO 17 and GPIO 22 as you wrote. Unfortunately I can't find yours
weh002004 and winstar_weh file (https://github.com/dhrone/pydPiper) under Displays / only the files winstar_weh.py and ws0010_test.py can be found.
Could you write me step by step what I have to do? I know I ask a lot of you, I would like to give you something for your trouble .....

Thank you so much for your help.
Greeting,
Karsten

dhrone, 2020-11-22:
Which music distribution are you using?

Karsten66, 2020-11-23:
Hello, 
i use Volumio 
the latest version, with the Winstar 4x20 - WEH002004A via parallel interface.
In all relevant programs I changed the GPIOS to 17 and 22. Something can only be seen in two lines on the OLED, but nothing real.
I found the weh002004 and winstar_weh.py files and copied them to the Displays directory, but nothing happened. 
I also used the pages_lcd_20x4.py file, modified to the GPIOS I am using.
Can you please help?

Greeting,
Karsten

Karsten66, 2020-11-23:
**Here is my software/hardware:**
My installation:
1.) sudo wget https://github.com/dhrone/pydPiper/archive/v0.3-alpha.tar.gz
2.) sudo tar zxvf v0.3-alpha.tar.gz
3.) sudo nano config.service

driver: /pydPiper/displays/weh002004/winstar_weh.py
musicdistro: Volumio
e   = 22
rs  = 17
d4 = 25
d5 = 24
d6 = 23
d7 = 27
Time zone: Germany / Berlin
Swap file: pages_lcd_20x4.py

4.) sudo ./install.sh
5.) sudo reboot

Display: WINSTAR OLED Display WEH002004A 20x4
Volumio: Version 2.853 (20-11-2020), 2.698 (16-01-2020), 2.692 (19_12-2019)
Sound card: Audiophonics DAC I-Sabre ES9038Q2M / I2S
Raspberry Pi 4, Model B, 4 GB RAM

greeting,
Karsten

Karsten66, 2020-11-23:
Hallo,
here my pydpiper.service file. Is everything right there?

# HD44780 LCD 20x4 display with parallel interface using the following pin assignments
# Enable - pin 22
# R/S    - pin 17
# D4     - pin 25
# D5     - pin 24
# D6     - pin 23
# D7     - pin 27

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py --volumio --driver winstar_weh --width 100 --height 32 --rs 17 --e 22 --d4 25 --d5 24 --d6 23 --d7 27 --timezone 'Germany/Berlin' --temperature celsiust --pages pages_lcd_20x4.py

Greeting,
Karsten

dhrone, 2020-11-23:
Ok.  You need to pull the current code, not the released version.  I've created a branch of the project specifically to work on the weh002004 driver.  So, here is a step-by-step for Volumio that assumes you have already gotten docker set up correctly.

After getting yourself logged in...

```
mkdir /home/volumio/weh
cd /home/volumio/weh
git clone https://github.com/dhrone/pydPiper
cd pydPiper
git checkout weh002004
```

This will pull down the latest version of the pydPiper code and switch you into the branch of the code that contains the new weh driver.  You will then need to run configure to set up the pydPiper system.

`python configure.py`

You should choose winstar_weh as your display type and set the RS, E and data pins consistent with how you have wired the display.  Choice of pageFile is up to you but keep in mind that the winstar_weh driver is character-based which limits how much graphical content it can display at any one time.  I would recommend starting from one of the LCD page files such as pages_lcd_20x4.py.

To test the installation you can run pydPiper from the command line.

`docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/weh/pydPiper:/app:rw dhrone/pydpiper:v0.32-alpha python /app/pydPiper.py`

You can exit using ctrl-C.

Assuming everything worked correctly, here is what you would put in your pydpiper.service file.

`
ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/weh/pydPiper:/app:rw dhrone/pydpiper:v0.32-alpha python /app/pydPiper.py
`

Note that all of the settings you used to specify using command-line flags have been moved to the pydPiper.cfg file which is created by configure.py.

Let me know if this tests correctly so that I can merge it into the main code branch.

dhrone, 2020-11-23:
Also, I've been working to move all of the display logic over to the luma.oled/luma.lcd projects.  So, if we get this working correctly here, I'd like to add the functionality to that project.  I do not however own a winstar weh002004 display for testing.  If I can provide instructions and a test script to you, would you be willing to perform the testing?

Karsten66, 2020-11-24:
Hello, 
if I can help - gladly.

Greeting,
Karsten

Karsten66, 2020-11-24:
Hello, do I have to install Docker separately? 
I thought it would be installed automatically.
Greeting,
Karsten

dhrone, 2020-11-24:
There is a script in the pydPiper root directory called install_docker.sh you can run.  Our you can just issue the following commands...

```
echo 'Installing Docker from the convenience script\n'
curl -sSL https://get.docker.com | sh

echo 'Enabling docker\n'
sudo systemctl enable docker
sudo systemctl start docker
sudo docker pull dhrone/pydpiper:v0.32-alpha

echo 'Testing docker\n'
sudo docker run hello-world
```

Karsten66, 2020-11-24:
It does not work.

Greeting,
Karsten

dhrone, 2020-11-24:
I'll need you to be a bit more specific.

Karsten66, 2020-11-24:
Install Volumio Version 2.698
Install Volumio Plugins
Install Docker:	

curl -sSL https://get.docker.com | sh

sudo systemctl enable docker

sudo systemctl start docker

sudo docker pull dhrone/pydpiper:v0.31-alpha

sudo docker run hello-world

sudo apt install python3
sudo apt install python3
sudo apt install python-mpd
sudo pip3 install mpdlcd

mkdir /home/volumio/weh
cd /home/volumio/weh
git clone https://github.com/dhrone/pydPiper
cd pydPiper
git checkout weh002004

cd /weh/pydPiper/
      phyton configure.py
      Startup…(5)		5
      Display typ		winstar_weh
      With…. Pixels		100
      Height… Pixels		60
      Enable…			1
      Register… pin		7
      Enable pin		8
      Data 4			25
      Data 5			24
      Data 6			23
      Data 7			27
      Location…		pages_lcd_20x4.py
      Animation…		0.1
      Location….		/var/log/pydPiper.log
      Logging…
      Time…
      24-hour…
      Temp…
      Enable…
      Name of…		volumio
      Server…			192.168.2.1 (Router???)
      Port…
      
      
      
And only what can be seen in the picture.

Greeting,
Karsten
      
      
      
      
				

Von: dhrone
Gesendet: Dienstag, 24. November 2020 18:09
An: dhrone/pydPiper
Cc: Karsten66; Comment
Betreff: Re: [dhrone/pydPiper] Problem with WINSTAR OLED WEH002004AR (20x4,RED) (#71)

I'll need you to be a bit more specific.

dhrone, 2020-11-24:
Sorry, I'm not following.  Let's go through the steps.

Did docker install correctly? Re-run...
`sudo docker run hello-world`
What output do you see?  It should start with the message 'Hello from Docker!'

Assuming docker is running lets check to see if you can access the container.  To run the container in interactive mode use...

`docker run -ti --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/weh/pydPiper:/app:rw dhrone/pydpiper:v0.32-alpha sh`

This should place you at a new command prompt that looks like this.
`/app # `

Now we can check to see if the driver is working correctly.

`python displays/winstar_weh --rs 17 --e 22 --d4 25 --d5 24 --d6 23 --d7 27`

This should cause the display to show:
```
Winstar WEH LCD
Pi Powered
```

Now you need to run the configure program.
`python configure.py`

Here is a printout of the answers I think you will need...
```
Creating configuration file for pydPiper
----------------------------------------

STARTUP CONFIGURATION

Startup mode duration (in seconds)? [5]: 

DISPLAY CONFIGURATION

Display type?: winstar_weh
Width of display (in pixels)? [80]: 100
Height of display (in pixels)? [16]: 32
Enable pulse duration (in microseconds)? [1]: 
Register select pin? [7]: 17
Enable pin? [8]: 22
Data 4 pin? [25]: 
Data 5 pin? [24]: 
Data 6 pin? [23]: 
Data 7 pin? [27]: 
Location of the pagefile? [pages_lcd_16x2.py]: pages_lcd_20x4.py
Animation Smoothing (in seconds)? [0.15]: 

SYSTEM CONFIGURATION

Location of log file? [/var/log/pydPiper.log]: 
Logging Level? [info]: 
Time Zone? [US/Eastern]: Germany/Berlin
24-hour clock? [FALSE]: 
Temperature Scale? [fahrenheit]: celsius
Enable weather? [n]: 

MUSIC DISTRIBUTION

Name of distribution?: volumio
Server address? [localhost]:          
Port? [3000]: 

UPDATING pydPiper.cfg
Creating pydpiper.service file
```

And next run pydPiper at the command line to see if it works correctly.

`python pydPiper.py`

Hit <Ctrl-C> to exit

To exit the docker container type:
`exit`

Let me know whether you get through all of this and we can proceed based upon your experience.

Karsten66, 2020-11-25:
Good morning, 
I did the following: installed new Volumio, installed Soundcard driver, plugin for the touchscreen.
Now Docker installed:

curl -sSL https://get.docker.com | sh

sudo systemctl enable docker

sudo systemctl start docker

sudo docker pull dhrone/pydpiper:v0.31-lpha

sudo docker run hello-world
Unfortunately nothing happens on the Winstar display. I left GPIO07, GPIO08 on pin 26 and 24.
Now I'm trying with a soundcart and GPIO17 and GPIO22.
Unfortunately, nothing happens on the OLED display.

Greeting,
Karsten

Von: dhrone
Gesendet: Dienstag, 24. November 2020 18:58
An: dhrone/pydPiper
Cc: Karsten66; Comment
Betreff: Re: [dhrone/pydPiper] Problem with WINSTAR OLED WEH002004AR (20x4,RED) (#71)

Sorry, I'm not following. Let's go through the steps.
Did docker install correctly? Re-run...
sudo docker run hello-world
What output do you see? It should start with the message 'Hello from Docker!'
Assuming docker is running lets check to see if you can access the container. To run the container in interactive mode use...
docker run -ti --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/weh/pydPiper:/app:rw dhrone/pydpiper:v0.32-alpha sh
This should place you at a new command prompt that looks like this.
/app # 
Now we can check to see if the driver is working correctly.
python displays/winstar_weh --rs 17 --e 22 --d4 25 --d5 24 --d6 23 --d7 27
This should cause the display to show:
Winstar WEH LCD
Pi Powered
Now you need to run the configure program.
python configure.py
Here is a printout of the answers I think you will need...
Creating configuration file for pydPiper
----------------------------------------

STARTUP CONFIGURATION

Startup mode duration (in seconds)? [5]: 

DISPLAY CONFIGURATION

Display type?: winstar_weh
Width of display (in pixels)? [80]: 100
Height of display (in pixels)? [16]: 32
Enable pulse duration (in microseconds)? [1]: 
Register select pin? [7]: 17
Enable pin? [8]: 22
Data 4 pin? [25]: 
Data 5 pin? [24]: 
Data 6 pin? [23]: 
Data 7 pin? [27]: 
Location of the pagefile? [pages_lcd_16x2.py]: pages_lcd_20x4.py
Animation Smoothing (in seconds)? [0.15]: 

SYSTEM CONFIGURATION

Location of log file? [/var/log/pydPiper.log]: 
Logging Level? [info]: 
Time Zone? [US/Eastern]: Germany/Berlin
24-hour clock? [FALSE]: 
Temperature Scale? [fahrenheit]: celsius
Enable weather? [n]: 

MUSIC DISTRIBUTION

Name of distribution?: volumio
Server address? [localhost]:          
Port? [3000]: 

UPDATING pydPiper.cfg
Creating pydpiper.service file
And next run pydPiper at the command line to see if it works correctly.
python pydPiper.py
Hit to exit
To exit the docker container type:
exit
Let me know whether you get through all of this and we can proceed based upon your experience.

Karsten66, 2020-11-25:
Good morning, 
I have reinstalled Volumio in one go (version 2.834). 
As far as all plugins are installed. 
Then Phyton and Docker:
cd
sudo apt install python3
sudo apt install python3
sudo apt install python-mpd
sudo pip3 install mpdlcd

curl -sSL https://get.docker.com | sh
sudo systemctl enable docker
sudo systemctl start docker
sudo docker pull dhrone/pydpiper:v0.31-alpha
sudo docker run hello-world

nothing...

docker run -ti --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/weh/pydPiper:/app:rw dhrone/pydpiper:v0.32-alpha sh

mkdir /home/volumio/weh
cd /home/volumio/weh
git clone https://github.com/dhrone/pydPiper
cd pydPiper
git checkout weh002004

python configure.py

docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/weh/pydPiper:/app:rw dhrone/pydpiper:v0.32-alpha pytthon /app/pydPiper.py

Nothing…

sudo nano pydpiper.service

In……

Here everything ...

login as: volumio
volumio@192.168.2.30's password:
                       ___
                      /\_ \                        __
         __  __    ___\//\ \    __  __    ___ ___ /\_\    ___
        /\ \/\ \  / __`\\ \ \  /\ \/\ \ /' __` __`\/\ \  / __`\
        \ \ \_/ |/\ \L\ \\_\ \_\ \ \_\ \/\ \/\ \/\ \ \ \/\ \L\ \
         \ \___/ \ \____//\____\\ \____/\ \_\ \_\ \_\ \_\ \____/
          \/__/   \/___/ \/____/ \/___/  \/_/\/_/\/_/\/_/\/___/

             Free Audiophile Linux Music Player - Version 2.0

          C 2015 Michelangelo Guarise - Volumio Team - Volumio.org

Volumio Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
volumio@volumio:~$ sudo apt install python3

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

[sudo] password for volumio:
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package python3
volumio@volumio:~$ sudo apt install python3
Reading package lists... Done
[... 222 lines cut ...]
Animation Smoothing (in seconds)? [0.15]: 

SYSTEM CONFIGURATION

Location of log file? [/var/log/pydPiper.log]: 
Logging Level? [info]: 
Time Zone? [US/Eastern]: Germany/Berlin
24-hour clock? [FALSE]: 
Temperature Scale? [fahrenheit]: celsius
Enable weather? [n]: 

MUSIC DISTRIBUTION

Name of distribution?: volumio
Server address? [localhost]:          
Port? [3000]: 

UPDATING pydPiper.cfg
Creating pydpiper.service file
And next run pydPiper at the command line to see if it works correctly.
python pydPiper.py
Hit to exit
To exit the docker container type:
exit
Let me know whether you get through all of this and we can proceed based upon your experience.

Karsten66, 2020-11-25:
Hello,
maybe i installed the wrong python ??? 
I have already reinstalled everything 6 times today, I must have made a mistake somewhere! 
Now the display also remains dark.

Greeting,
Karsten

Von: dhrone
Gesendet: Dienstag, 24. November 2020 18:58
An: dhrone/pydPiper
Cc: Karsten66; Comment
Betreff: Re: [dhrone/pydPiper] Problem with WINSTAR OLED WEH002004AR (20x4,RED) (#71)

Sorry, I'm not following. Let's go through the steps.
Did docker install correctly? Re-run...
sudo docker run hello-world
What output do you see? It should start with the message 'Hello from Docker!'
Assuming docker is running lets check to see if you can access the container. To run the container in interactive mode use...
docker run -ti --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/weh/pydPiper:/app:rw dhrone/pydpiper:v0.32-alpha sh
This should place you at a new command prompt that looks like this.
/app # 
Now we can check to see if the driver is working correctly.
python displays/winstar_weh --rs 17 --e 22 --d4 25 --d5 24 --d6 23 --d7 27
This should cause the display to show:
Winstar WEH LCD
Pi Powered
Now you need to run the configure program.
python configure.py
Here is a printout of the answers I think you will need...
Creating configuration file for pydPiper
----------------------------------------

STARTUP CONFIGURATION

Startup mode duration (in seconds)? [5]: 

DISPLAY CONFIGURATION

Display type?: winstar_weh
Width of display (in pixels)? [80]: 100
Height of display (in pixels)? [16]: 32
Enable pulse duration (in microseconds)? [1]: 
Register select pin? [7]: 17
Enable pin? [8]: 22
Data 4 pin? [25]: 
Data 5 pin? [24]: 
Data 6 pin? [23]: 
Data 7 pin? [27]: 
Location of the pagefile? [pages_lcd_16x2.py]: pages_lcd_20x4.py
Animation Smoothing (in seconds)? [0.15]: 

SYSTEM CONFIGURATION

Location of log file? [/var/log/pydPiper.log]: 
Logging Level? [info]: 
Time Zone? [US/Eastern]: Germany/Berlin
24-hour clock? [FALSE]: 
Temperature Scale? [fahrenheit]: celsius
Enable weather? [n]: 

MUSIC DISTRIBUTION

Name of distribution?: volumio
Server address? [localhost]:          
Port? [3000]: 

UPDATING pydPiper.cfg
Creating pydpiper.service file
And next run pydPiper at the command line to see if it works correctly.
python pydPiper.py
Hit to exit
To exit the docker container type:
exit
Let me know whether you get through all of this and we can proceed based upon your experience.

mariuszmysz, 2020-11-25:
Hi,
Observing your discussion, I decided to cut in.
@Karsten66 - I have a question for you: have you tested your display independently of the @dhrone firmware?
If so, ignore the rest of my message.

If not:

- forget the pydpiper for a moment
- see the script of the robertcx user (https://www.raspberrypi.org/forums/viewtopic.php?t=68055)
- copy the last script of this user and run on your raspberry - it must display a test two-line information
- this will tell you if the physical connection of your raspberry and display is correct - if you can't display anything, don't install pydpiper
- make sure that in the test script you mapped the display pins to the raspberry pins in the same way as for the pydpiper - this is important because you can go straight to the pydpiper test from the simple test

Comments:

- Winstar displays are ungrateful devices and the tech support is rudimentary (I got some, but I would have to be an embeded developer to deal with it); The @dhrone has done an amazing job to make it easy to install and connect various displays, but be mentally prepared for problems that you wouldn't encounter with ordinary LCDs; if you go through it, you will enjoy it because the effect is nice; personally many times I wanted to give up already, but managed to connect;

- make sure your display works with 3V3 and can tolerate the logic levels of the raspberry

- not all display Winstar models can work with 3V3 - then logic level converters must be used

These displays, however, have an advantage that captivated me - the perfect contrast.

English is not my native language, I use a translator - I could have made grammatical mistakes because I like sentences that are complex many times - so I apologize in advance :)

Best regards
@mariuszmysz

dhrone, 2020-11-25:
Thanks @mariuszmysz.  That is good advice.

Karsten66, 2020-11-26:
Hello,
which version of Python does Volumio need and how is the installation?
Greeting,
Karsten

mariuszmysz, 2020-11-26:
Hi,
the precise answer to your question is: it doesn't need any version.
Volumio is a ready image of the operating system that contains the necessary components to start to perform its functions.
But my guess is where your question came from: you probably want under Volumio image to run python scripts.

So after booting the system and connecting to the console, check if you have any version installed at all. Here is the description:

https://note.nkmk.me/en/python-sys-platform-version-info/

If you don't have any version - find some article on how to install the python interpreter.

As for the tests - the python version is not important as the display test scripts are unlikely to use version dependent commands.

You will probably need the RPI.GPIO module. He is here:

https://sourceforge.net/p/raspberry-gpio-python/wiki/install/

This is a frequently used module that supports interaction with the Raspberry GPIO.

After testing the display and before installing the pydpiper, see this article:

https://en.wikipedia.org/wiki/Docker_(software)

Try to read it a few times with understanding and remember that this "something" exists.
This technology is based on how the pydpiper starts. You will probably never need to use Docker, but you should know where the word "docker" comes from.
You will slowly start to understand everything :)

Best regards
Marius

Karsten66, 2020-11-26:
Good morning Marius,
Thank you very much for your answer, I didn't know that you actually didn't need Python, thank you very much. 
Unfortunately, yesterday I didn't get your tip to test the "robertcx script".
**Again, many thanks to you guys who help me so nicely! You are just great !!!**
You do a great job!

Greeting,
Karsten

Karsten66, 2020-11-26:
Hello Marius, 
could you help me whether my installation is so correct?

**Install Volumio (Version 2.853)**

**Install_docker.sh**
1.)	curl -sSL https://get.docker.com | Sh
2.)	sudo systemctl enable docker
3.)	sudo systemctl start docker
4.)	sudo docker pull dhrone/pydpiper:v0.31-alpha
5.)	sudo docker run hello-world

**Install_volumio.sh**
6.)	sudo python configure.py
7.)	sudo cp pydpiper.service /etc/systemd/system/pydpiper.service
8.)	sudo systemctl enable pydpiper
9.)	sudo systemctl start pydpiper

**Install pydPiper**
10.)	sudo wget https://github.com/dhrone/pydPiper/archive/v0.3-alpha.tar.gz
11.)	sudo tar zxvf v0.3-alpha.tar.gz
12.)	cd pydPiper-0.3-alpha
13.) 	sudo nano pydpiper.service 

This worked for a 20x4 I2C LED display.
**pydpiper.service** - I2C LCD 20x4-Display
# HD44780 LCD 20x4 display with I2C interface on address 0x27

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py --volumio --driver hd44780_i2c  --width 100 --height 32 --i2caddress 0x27 --timezone 'Europe/Berlin' --temperature celsius --pages pages_lcd_20x4.py

**Where must the values for the location / 24h display and temperature be changed? 
Only in the pydpiper.service?**

And for a Winstar OLED display with a parallel interface, this?
**pydpiper.service** - Winstar 20x4 parallel interface OLED Display

# HD44780 LCD 20x4 display with parallel interface using the following pin assignments
# Enable	- pin 25
# R/S    	- pin 24
# D4     	- pin 23
# D5     	- pin 17
# D6     	- pin 27
# D7     	- pin 22

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py --volumio --driver winstar_weh --width 100 --height 32 --rs 7 --e 8 --d4 25 --d5 24 --d6 23 --d7 27 --timezone 'Europe/Berlin' --temperature celsius --pages pages_lcd_20x4.py

Greeting,
Karsten

mariuszmysz, 2020-11-26:
Hi, Karsten

Before I answer, I would like to make sure of two things:

- did you perform a simple test without installing pydpiper as described in an earlier message?

 - _This worked for a 20x4 I2C LED display._  - does it mean that you have started the display on the I2C bus and is already showing something?

Karsten66, 2020-11-26:
Hello Marius,
my display 20x4 LED with I2C interface works without a problem on my Raspberry Pi 4 with Volumio.
The same system, only with the new Volumio and Winstar OLED 4x20 display, without all, the test shows nothing in the display.

A completely different scenario, Volumio with pydPiper, Docker and the Winstar OLED 4x20, the display only shows strange characters.

mariuszmysz, 2020-11-26:
> Hello Marius,
> my display 20x4 LED with I2C interface works without a problem on my Raspberry Pi 4 with Volumio.
> The same system, only with the new Volumio and Winstar OLED 4x20 display, without all, [...]

ok,

therefore:

I2C + LCD + without pydpiper -> working
I2C + OLED -> no pydpiper -> not working, empty display
I2C + OLED + pydpiper> not working -> strange characters

?

Karsten66, 2020-11-26:
I2C + LCD 20x4 with Pydpiper works. 
OLED 20x4 Winstar with serial port + Pydpiper strange characters.
OLED 20x4 Winstar with serial port not working.

mariuszmysz, 2020-11-26:
> 
> 
> I2C + LCD 20x4 with Pydpiper works.
> OLED 20x4 Winstar with serial port + Pydpiper strange characters.
> OLED 20x4 Winstar with serial port not working.
> Is my installation routine correct?

If you managed to get LCD + I2C running - then you are on the right way.

However, when it comes to serial connection I can't support you because I don't have enough experience with serial connection.
If you care a lot about serial connection, you should continue with the test described by the @dhrone :

> Sorry, I'm not following. Let's go through the steps.
>
> Did docker install correctly? Re-run...
> `sudo docker run hello-world`
> What output do you see? It should start with the message 'Hello from [...]

**The strange characters** result from an incorrect sequence of instructions that initialize the display. This is due to the fact that WINSTAR only declares that its displays conform to the hd44780 standard.

In fact, they require a different initialization sequence than regular LCDs. That's why the LCD works for you, and OLED shows "garbage".

@dhrone has prepared its software in accordance with hd44780 so OLED does not want to work.
He was kind enough to prepare a driver for you that might be able to handle an OLED display.
It is now up to you to decide whether you want to continue this test.
I have experience in the area of ​​OLED + I2C - as you probably read at the beginning.

Earlier you wrote that you have some problem with displaying degrees Celsius and the 24 hour format.
Once you deal with starting the display, contact me and I will try to explain to you how to change it.

best regards
Marius

Karsten66, 2020-11-27:
Hello Marius,
late yesterday evening I try to install only the Docker on a new Volumio without plugins and sound card, only the Winstar 20x4 OLED display, of course without success.
I entered the following:
-curl -sSL https://get.docker.com | sh
-sudo systemctl enable docker
-sudo systemctl start docker
-sudo docker pull dhrone/pydpiper:v0.31-alpha
-sudo docker run hello-world
Hello world did not appear on the display and there was an error message that the file hello-world could not be found.
Is it maybe my Raspberry Pi 4 to blame, or the Volumio version 2.853?
With my LCD display 4x20 hd44780 I2C, I can only get everything to work if I am using an older version and with python. 
I think because otherwise he cannot start the written programs (....py).
I'm just at a loss!
I'm already thinking about whether I should send the OLED to you with a Raspberry, sound card, money to send back and everything else that goes with it. What is needed for the display to run? 
1.Volumio,
2.Docker,
3.
Could you describe it to me as a stupid user, piece by piece? I connected my display like this, maybe it collides with some GPIO?
PIN1 from the OLED=GND, 
PIN2 =+5V, 
PIN3 =not connected, 
PIN4 (RS) =GPIO7, other GPIO17, other GPIO25, 
PIN5 (R/W) =GND, 
PIN6 (E.) =GPIO8, other GPIO22, other GPIO24,
PIN7-10 =not connected, 
PIN11(D4) =GPIO25,  GPIO25, other GPIO23.
PIN12(D5) =GPIO24,  GPIO24, other GPIO17.
PIN12(D6) =GPIO23,  GPIO23, other GPIO27.
PIN13(D7) =GPIO27   GPIO27, other GPIO22.
PIN14-15 =not connected.
Would that be okay?
I'm really sorry that I annoy you so much, but I want to finally have a success. The display also cost more than € 45.
Many, many thanks again to everyone who helps me.
Best regards,
Karsten

mariuszmysz, 2020-11-27:
Hi, @Karsten66 

I will try to support you, as long as you follow my recommendations precisely.
Forget what I've done so far and do the following:

- put a fresh image of the latest volumio
- do not connect any devices to the raspsberry
- execute the command:

"$ python --version

Send me the result of this instruction, do not take any other action.
When implementing volumio and pydpiper you have to deal with two types of bugs: software and hardware. First, we will deal with software.

Marius

Karsten66, 2020-11-27:
Great, thank you. So I'll try it out afterwards and then write.
Greeting,
Karsten

Karsten66, 2020-11-28:
Hallo Marius,
I entered python --version,
Python 2.7.9 is displayed.
what's next
greeting,
karsten

mariuszmysz, 2020-11-28:
Hi, Karsten

follow these steps:

1. prepare stereo headphones (3-pole jack) or a set of computer speakers; run any Internet radio station - you should hear a sound through the speakers, then follow these instructions:

`volumio @ volumio: ~ $ volumio status`

send me the result.

2. remember that you are to familiarize yourself with what a docker is and follow these instructions:

`volumio@volumio: ~ $ sSL get.docker.com -o get-docker.sh && sh get-docker.sh`

`volumio@volumio: ~ $ sudo gpasswd -a $ USER docker`

`volumio@volumio: ~ $ newgrp docker`

`volumio@volumio: ~ $ docker run hello-world`

send me the result of the **last** instruction.

Marius

Karsten66, 2020-11-28:
[image]
[image]

Karsten66, 2020-11-28:
[image]

mariuszmysz, 2020-11-28:
I felt something like this ...

if I am writing you to follow the instruction:

`volumio@volumio:~$ volumio status`

then the instruction is everything after the $ sign, that is:

`volumio status`

come on, such things you need to know, look at your command line ...

Karsten66, 2020-11-28:
[image]

Karsten66, 2020-12-18:
Hello my dears,
I finally did it and now my Winstar OLED 20x4 shows everything. I still have to fathom how I did it, I just had everything installed and at some point it worked. Excellent! Thank you again for your patience with me. But now to a request, Marius. You wanted to help me to see the 24 hour clock format and Celsius on the display, what should I do?
Greeting,
Karsten

mariuszmysz, 2020-12-20:
Hi Karsten,

I encountered the problem of displaying in 24H format personally. In my case, the contents of the PAGES file had to be changed.

**Before making any changes, always back up your files or create an image of the SD card to have a working system.**
How it works and what is the role of the PAGES _ *. Py file I will not describe because @dhrone has already described it.

During the service setup, you selected a display driver and its associated PAGES file. This file contains a format string for the displayed time. I will describe it based on my own file. I use the TIME widget to display the time, so find this line in the PAGES file (I'm giving a fragment, not literally):

`'time': {'type': 'text', 'format': '{0}', 'variables': ['localtime | strftime +% - I:% M'].`

and change the format string from:

`localtime|strftime+%-I:%M`
on
`localtime|strftime+%H:%M`

It is possible that there is an AMPM widget in your PAGES file and it is used in the CANVASES section - if so - maybe you should change the formatting for it too.
Note: after changing the formatting, you may find that the time is displayed incorrectly - in this case, try changing parameters in the WIDGETS section, such as:

`just`
`varwidth`
`size`

Here are my own settings, I have a 4x20 display and use the pages_lcd_20x4 file:

WIDGTES:
`'time': {'type': 'text', 'format': '{0}', 'variables': [' localtime | strftime +% H:% M '],' font ':' large ',' just ':' right ',' varwidth ': True,' size ':( 65,16)}`

CANVASES
`'stoptime': {'widgets': [('time', 15.8)], 'size' :( 100.32)}`

In addition to editing the PAGES file, perform the service configuration again and on the 24HOUR line enter exactly **True** (in upper case).

Always restart the pydpiper service after making changes.
To make sure that you are editing the correct PAGES file, try to make a minor change, for example: replace lowercase letters in the "splash" widget with uppercase and restart the service - you should see the changes on the display.

Regarding degrees Celsius: however, in my notes I found no description of any problem. Therefore, call the configuration again and in the question about the degrees, enter celsius and make sure that you do not make a mistake in this name.
If that doesn't work, you need to contact @dhrone.

Marius

Karsten66, 2021-01-15:
Hello Marius,
I wish you a happy, healthy new year. I hope you are alive and well.

I haven't been able to solve the problem with the 24H-Format until today, but I'm staying on the ball.

But now I have a new question, is it possible or is there already a suitable page for my 20x2 VFD Noritake display?
At the moment I'm using pages_lcd_20x4.py or pages_lcd_16x4, but that's not a solution, but I would like to have a suitable one.

Karsten

mariuszmysz, 2021-01-17:
Hi, Karsten

I'm still alive, but it is known for a time, happy new year :)

Your problem seems simple to solve as long as your Noritake display has a Hitachi HD44780 compatible driver: make a copy of pages_lcd_20x4 and name it pages_lcd_20x2 for example.

In this file you need to resize the widgets to fit the 20x2 display (by trial and error - note that the original size is 100 points wide and 32 points high).

In the same file you have to define the new position of the widgets on the individual canvases to match the new display height (16 points). For this work, follow the example of pages_lcd_16x2.

When you prepare the file, you need to call the pydpiper service configuration again and specify: new display size, i.e. 100x16 points, enter a new driver: hd44780 or hd44780_i2c (I assume that the Noritake model is character - not graphic) and as the PAGES file, enter the new prepared pages_lcd_20x2.

I can't guarantee this approach will work, but if I had a problem like you described, I would start there. I can't check it because I don't have a working test environment.
A long time ago I did such tests with a 16x2 display and it seems to me that it worked.

Marius

Karsten66, 2021-01-21:
Hello Marius,
my current time is now finally displayed in 24H.
That was really easy, I always edited the wrong pages_lcd_20x4.py. But now everything works.
Where I still find it difficult is the position of the clock, as I would like to see the seconds as well.
To write my own pages for my HD44780 i2c 20x2 display from Noritake, I still need a lot, a lot of time, for me as a beginner it's really difficult!
Despite everything, I am happy with your software, a huge thank you again to everyone.
Greeting,
Karsten

closed 2021-01-21 by dhrone

mariuszmysz, 2024-04-27:
Hi, hcanning87

After four years, I can only help you theoretically. I no longer have devices. I also abandoned the project because the OLED was too unstable. The LCD worked OK, but I really wanted to support OLED which looks cool. We shouldn't write here because this issue is closed by the author and it will be difficult for anyone to find this information. However, since I spent many days researching, I understand why you decided to write.

I don't do display programming at the moment, but I'll try to help you.
During that time, I tested various configurations to rule out where the problem was and where I was making a mistake. You should also realize that the drivers of these displays are not compatible with HD44780 - if they were, they would work as stably and predictably as LCD. My description assumes that your knowledge of LCD programming is at a level that allows you to understand, among others: such as the display initialization procedure.

If I understand correctly, you have this problem: restarting the service spoils the display, but restarting along with disconnecting the display power removes the problem.
Moreover, the problems with this manufacturer's OLED display can be summarized as follows: they are very sensitive to an incorrect sequence of instructions, understood as a sequence in an incorrect order in relation to the CURRENT state of the display, and sensitive to the delays between subsequent instructions. Incorrect sequences of instructions cause the display to be in a state in which, for example, it displays "garbage", does not display what you want, and the cursor may be in a random place, which causes the display to start from a random column. Unfortunately, there is no easy way to restore the display memory to its proper state - this can only be done by disconnecting the power supply. However, this is the worst possible solution.

I assume that in your case there is a problem in two places: when closing the pydPiper service: it terminates, which puts the display in an unstable state, the display still has power so its unstable state persists, then the service starts (second place) and tries to initialize it again – however, the initialization fails due to the fact that the OLED is in such a state that this is not possible.

First, you need to rule out a display failure. If you haven't done this yet, you can do it by writing your own code in python that initializes the display and displays something in a loop and does it long enough to assess that the display is working and the code that handles it is correct. This should be the code that controls the display connected as you want it to be connected to RPI and pydPiper.

Next, you need to analyze the code of the HD44780 control script and compare its initialization sequence with the template script (which always works), find the differences and explain them. Unfortunately, this is hard work that takes away motivation. I won't help you much remotely (too many variables), you need to have the device connected in the workshop and test it in various configurations. Delay functions are very important and may work differently in a system such as RPI - as you can see, in my case, changing to delay handling using time.sleep() helped.

It should be taken into account that there may be an error somewhere in the pydPiper code (it is unlikely, but it should be taken into account) that causes incorrect operation for this specific display (2x40). My display is 20x4. By the way, it was beautiful, red, it was supposed to work behind a red plexiglass filter, it was supposed to be great, but in the end it didn't work out.
Interestingly, I even asked Winstar technical support and received sample code in C, but I wasn't able to achieve much based on it. Even users on the Internet are divided on how these displays should be controlled. I also tested the connection with AVR ATMEGA, it worked, but that was not the point.

I don't know if I helped you with this long description, but best regards

Marius

hcanning87, 2024-04-27:
Hi Marius,

Thanks so much for your detailed reply. It’s a shame you abandoned it -
what did you find unstable about your OLED setup? Mine seems to work
brilliantly once it has been power cycled and it’s up and running. I just
need to try some of the initialisation / reset commands to make it survive
a soft reboot.

I too love how the OLEDs look compared to LCDs so I’m going to persevere :)

Were you connected via i2c?

Thanks again!