dhrone/pydPiper #72: Problems with 128x32 OLED

j0mar0, 2019-10-01:
Hi dhrone and thank you so much for making pydPiper available.
I am trying to get a 128x32 OLED to run with pydPiper but without success.

After a fresh install of Volumio 2.619 and pydPiper with 
git clone https://github.com/dhrone/pydPiper
cd pydPiper
sudo ./install_docker.sh
sudo ./install_volumio.sh

The cfg file has the following values:
[STARTUP]
startup_msg_duration = 5

[DISPLAY]
display_driver = luma_i2c
display_i2c_port = 1
display_i2c_address = 0x3c
display_devicetype = ssd1306
display_width = 128
display_height = 32
pagefile = pages_lcd_16x2.py
animation_smoothing = 0.15

[SYSTEM]
logfile = /var/log/pydPiper.log
loglevel = info
timezone = Atlantic/Azores
time24hour = true
temperature = celsius

[SOURCE]
source_type = volumio
volumio_server = localhost
volumio_port = 3000

And have nothing on the display.
Logfile has the following error:
2019-10-01 11:07:23,195:INFO:pydPiper starting
2019-10-01 11:07:30,613:INFO:Exiting...
2019-10-01 11:07:30,614:ERROR:Uncaught exception
Traceback (most recent call last):
  File "/app/pydPiper.py", line 746, in <module>
    lcd.update(img)
  File "/app/displays/luma_i2c.py", line 88, in update
    self.device.display(img)
  File "/usr/local/lib/python2.7/dist-packages/luma/oled/device.py", line 173, in display
    assert(image.size == self.size)
AssertionError
2019-10-01 11:07:30,619:ERROR:Player status at exception
2019-10-01 11:07:30,620:ERROR:{u'ip': u'10.10.108.45', u'random': 0, u'outside_temp_min': 0, u'channels': 0, u'single': 0, u'd$
2019-10-01 11:07:41,101:INFO:pydPiper starting
2019-10-01 11:07:48,513:INFO:Exiting...

After this if I rerun the "install volumio" script and change the display height to 64, the lcd display is shown as expected in a 128x64 display but squeezed in a 128x32 display.

Do I need to alter any other file?
Thanks for your help

j0mar0

monte-monte, 2019-11-23:
I've run into this issue too. The problem is here: https://github.com/dhrone/pydPiper/blob/aa1671011726fdec63e244d085ab95d085a21965/displays/ssd1306_i2c.py#L47
Code doesn't take into account the actual height and width which are passed as a parameters. And luma driver is initialized with default height of 32px.
It should be changed to `self.device = ssd1306(serial, cols, rows) `. Try yourself, it worked for me.

j0mar0, 2019-11-26:
Thank you so much for your answer and help, monte-monte, but I could not get any result.
For some reason I still got the same assertion error...

> 2019-11-26 11:33:42,640:ERROR:Uncaught exception
Traceback (most recent call last):
  File "/app/pydPiper.py", line 746, in <module>
    lcd.update(img)
  File "/app/displays/luma_i2c.py", line 88, in update
    self.device.display(img)
  File "/usr/local/lib/python2.7/dist-packages/luma/oled/device.py", line 173, in display
    assert(image.size == self.size)
AssertionError
2019-11-26 11:33:42,650:ERROR:Player status at exception
2019-11-26 11:33:42,651:ERROR:{u'ip': u'10.10.108.107', u'random': 0, u'outside_temp_min': 0, u'channels': 0, u'single': $

Again, if I rerun install_volumio.sh and set height of 64, the display comes back to life.
My limited knowledge of python does not permit me to find where the problem may be.
Is it the change in ssd1306_i2c.py the only change that is required?

Thank you again for your help.
j0mar0

monte-monte, 2019-11-26:
The problem is that you're using docker image, and it already contains ssd1306_i2c.py inside. So either maintainer of the docker image needs to update it, or you need to edit the file inside container yourself. 
Another option is to use pydpiper without docker. To do so you need to install required python libraries which you can find [here](https://github.com/dhrone/pydPiper/blob/master/requirements.txt).

I don't know if you are willing to to any of this, the easiest way would be to wait for @dhrone to update the code and docker image, then everything will work accordingly.

j0mar0, 2019-11-26:
Thank you for a quick reply, monte-monte. I've come across the problem of what is inside the docker container and had solved it as per issue 63 but than I started to install v0.31 to correct the time problem.
I'll have a look with my limited knowledge and try to solve according to your options.
If I find a way, i will post my findings here.
In the meantime maybe @dhrone has the time and opportunity to update the code.
Thank you so much for your help and guidance!

J0mar0