dhrone/pydPiper #78: Problem with 16x2 in parallel mode

tdouez, 2019-12-03:
Hello,
   I connected a generic 16x2 LCD on Raspberry / Volumio and after turning on pydPiper the display is incorrect with random characters.
   I tried with an LCD 20x4; i checked configuration and wiring; same issue ..

   
[image]
[image]

   Help ! :(
   Tristan

tdouez, 2019-12-03:
Sorry my problem is similar at issue #43

tdouez, 2019-12-05:
Hello,
    I bypassed this issue with an interface i2c/parallel.  :D 

Tristan

TaliesinDS, 2020-02-07:
I have the same issue.
Just random letters flickering on and off and occasionally one or more of the letters seem to represent the song that is playing but mostly just random garbage.

I don't think i can use an i2c interface since Hifiberry dac plus is using it and everywere i read it's not simple to do 2 things on i2c and i am a noob at this.

I'm running this on rpi 4 with a fresh image of volumio 2 and nothing else installed besides the pydpiper plugin.