dhrone/pydPiper #81: system crashes

manolomalele, 2019-12-20:
Hi dhrone and thank you really muck for pydPiper and for the easy setup.

I'm using a raspberry pi3b+ with moode. 

I've followed the instructions and everything works fine.
I use a 20x4 lcd display that shows all the info.
I also managed the widgets and the sequences in order to show on the display what i was looking for.

I tried to add in my display configuration the following widgets, taken from another config file:

`'bitdepth': { 'type':'text', 'format':'{0}', 'variables':['bitdepth'], 'font':'small', 'just':'center','varwidth':True},
'samplerate': { 'type':'text', 'format':'{0}', 'variables':['samplerate'], 'font':'small', 'just':'center','varwidth':True},
`

and then I've updated canvas, as usual.

But, once rebooted the system the docker doesn't work due to a problem related to the lined i've added.

The whole system failed, asking me for a new dac setup and the album art disappeared from the mail screen. Probably this affects the MPD. 

Is there a common issue about that?
Am I doing it wrong?

I really need those info (Bitrate/filetype/encoding) but i've no card to play.

Any helps or suggestions?

Thank you in advance

dhrone, 2019-12-31:
It's unclear what would be causing the crashing that you are describing just because you have added some additional variables to the WIDGETS definition.  A mistake in the pages file would normally only cause the pydPiper system to fail to start.  Where there any error messages that were displayed or logged?  

FYI, you should not be using the 'just':'center' parameter unless you are going to provide a specific 'size' parameter as pydPiper needs the size in order to determine what to justify to.  It is possible that there is a bug in that section of the code so you might try eliminating it.

manolomalele, 2020-01-20:
Hi Dhrone,
nothing to do...
I've changed the value as you suggested, but it doesn't work.
Do you have any idea how or what I have to add to my page_lcd_20x4 to make it work with those information?
Thank you in advance!

dhrone, 2020-02-15:
Please send me your page_lcd_20x4.py file and I'll see if I can locate what is causing the issue.

manolomalele, 2020-04-21:
Hi, sorry for the long time gone... I've changed my system. Now I use a ssd1306 oled wich works perfectly. I'm just wandering if is it possible to add, on the clock screen a line with the current ip address and a simple text like "this is my mediaplayer".
Thanks a lot

dhrone, 2020-04-22:
Yes, you'd need to modify the pages file that you are using to include a text (or ttext if using a truetype font) widget that uses the variable 'ip'.

Example:
in WIDGETS section...
`	'ipaddr': { 'type':'text', 'format':'{0}', 'variables':['ip'], 'font':'large', 'varwidth':True },
`

You would also need to place the new widget onto a canvas, presumably stoptime or stoptimetemp_popup if you are using one of my default page files.

See 'Pages file description.txt' inside of the docs folders for additional documentation.

manolomalele, 2020-04-24:
Thank you really much.

And what if I would simply add a text? 
Same thing on the widget paragraph? Sorry if these questions seem to be silly to you, but this my first approach with python .

Thanks again

dhrone, 2020-04-24:
You can create static text widgets the exact same way as widgets that use variables.  

`'statictext': { 'type':'text', 'format':'this is my media player', 'font':'large', 'varwidth':True },`

You can also combine text and variables:
`'name_address': { 'type':'text', 'format':'My Media Player using IP {0}', 'variables':['ip'], 'font':'large', 'varwidth':True },`

You can also use the newline symbol to make a textwidget use more than one line to display its content.
`'name_address': { 'type':'text', 'format':'My Media Player\nIP {0}', 'variables':['ip'], 'font':'large', 'varwidth':True },`

manolomalele, 2020-04-25:
Hi Dhrone, I've spent a couple of hours trying to figure it out, but I cannot understand the problem.
I've added the lines as you suggested. Then rebooted.
Nothing on the display.

Here it is the code:
pages_ssd1306.py
`#!/usr/bin/python.pydPiper
# coding: UTF-8

from __future__ import unicode_literals

# Page Definitions
# See Page Format.txt for instructions and examples on how to modify your display settings

# Load the fonts needed for this system
FONTS = {
        'small': { 'default':True, 'file':'latin1_5x8.fnt','size':(5,8) },
#       'large': { 'file':'BigFont_10x16.fnt', 'size':(10,16) },
        'large': { 'file':'Vintl01_10x16.fnt', 'size':(10,16) },
        'tiny': { 'file':'upperasciiwide_3x5_fixed.fnt', 'size':(5,5) },
}

TRUETYPE_FONTS = {
        'DejaVuSans28': { 'file':'DejaVuSans.ttf', 'size':28 }
}

IMAGES = {
        'progbar': {'file':'progressbar_80x8.png' },
        'splash': {'file':'pydPiper_fixed_splash.png' }
}

# Load the Widgets that will be used to produce the display pages
WIDGETS = {
        'xofy': { 'type':'text', 'format':'{0}/{1}', 'variables':['playlist_position', 'playlist_length'], 'font':'small', 'size':(58,16), 'varwidth':True },
        'volume': { 'type':'text', 'format':'Volume: {0}', 'variables':['volume'], 'font':'small', 'just':'right', 'size':(60,16), 'varwidth':True },
        'volumelarge': { 'type':'text', 'format':'Volume: {0}', 'variables':['volume'], 'font':'large', 'just':'left', 'varwidth':True },
        'volumebar': { 'type':'progressbar', 'value':'volume', 'rangeval':(0,100), 'size':(115,8) },
        'artist': { 'type':'text', 'format':'{0}', 'variables':['artist'], 'font':'large','varwidth':True,'effect':('scroll','left',1,1,20,'onloop',3,125)},
        'title': { 'type':'text', 'format':'{0}', 'variables':['title'], 'font':'small','varwidth':True,'effect':('scroll','left',1,1,20,'onloop',3,125)},
        'samplerate': { 'type':'text', 'format':'{0}', 'variables':['samplerate'], 'font':'small', 'just':'center','varwidth':True},
        'bitdepth': { 'type':'text', 'format':'{0}', 'variables':['bitdepth'], 'font':'small', 'just':'center','varwidth':True},
        'elapsed': { 'type':'text', 'format':'{0}', 'variables':['elapsed|strftime+%-M:%S'], 'font':'small','size':(30,8), 'varwidth':True},
        'length': { 'type':'text', 'format':'{0}', 'variables':['length|strftime+%-M:%S'], 'font':'small','size':(30,8),'just':'right','varwidth':True},
        'songprogress': { 'type':'progressbar', 'value':'elapsed', 'rangeval':(0,'length'), 'size':(62,8) },
        'time': { 'type':'text', 'format':'{0}', 'variables':['utc|timezone+Europe/Rome|strftime+%-I:%M'], 'font':'large', 'just':'right', 'varwidth':True, '$
        'ttime': { 'type':'ttext', 'format':'{0}', 'variables':['utc|timezone+Europe/Rome|strftime+%-I:%M'], 'font':'DejaVuSans28', 'just':'right', 'varwidth$
        'ampm': { 'type':'text', 'format':'{0}', 'variables':['utc|timezone+Europe/Rome|strftime+%p'], 'font':'tiny', 'varwidth':True },
        'tempsmall': { 'type':'text', 'format':'{0}', 'variables':['outside_temp_formatted'], 'font':'small', 'just':'right', 'size':(24,16) },
        'temphilow': { 'type':'text', 'format':'H {0}\nL {1}', 'variables':['outside_temp_max|int', 'outside_temp_min|int'], 'font':'small', 'just':'right', $
        'temp': { 'type':'text', 'format':'{0}', 'variables':['outside_temp_formatted'], 'font':'large', 'just':'center', 'size':(80,16) },
        'weather': { 'type':'text', 'format':'{0}', 'variables':['outside_conditions|capitalize'], 'font':'large','varwidth':True, 'effect':('scroll','left',$
        'ipaddr': { 'type':'text', 'format':'{0}', 'variables':['ip'], 'font':'large', 'varwidth':True },
        'statictext': { 'type':'text', 'format':'High Definition Media Player', 'font':'small', 'varwidth':True },
}

# Assemble the widgets into canvases.  Only needed if you need to combine multiple widgets together so you can produce effects on them as a group.
CANVASES = {
        'playing': { 'widgets': [ ('artist',3,10), ('title',3,26), ('bitdepth',22,45), ('samplerate',78,45), ('elapsed',3,56), ('length',98,56), ('songprogre$
        'stoptime': { 'widgets': [ ('ttime',10,20), ('ipaddr',20,56) ], 'size':(128,64) },
 'stoptimetemp_popup': { 'widgets': [ ('ttime',3,0), ('tempsmall',100,0), ('weather',8,47), ('temphilow',100,48) ], 'size':(128,64) },
        'volume_changed': { 'widgets': [ ('volumelarge',8,0), ('volumebar',8,18) ], 'size':(128,64) }
}

# Place the canvases into sequences to display when their condition is met
# More than one sequence can be active at the same time to allow for alert messages
# You are allowed to include a widget in the sequence without placing it on a canvas

# Note about Conditionals
# Conditionals must evaluate to a True or False resulting
# To access system variables, refer to them within the db dictionary (e.g. db['title'])
[... 35 lines cut ...]
    dc.load(pagefile, mc.musicdata,mc.musicdata_prev )
  File "/app/displays/display.py", line 1313, in load
    newpages = imp.load_source('pages', file)
  File "pages_ssd1306.py", line 47
    'ipaddr': { 'type':'text', 'format':'{0}', 'variables':['ip'], 'font':'large', 'varwidth':True },
           ^
SyntaxError: invalid syntax
Player status at exception
{u'ip': u'192.168.1.7', u'random': 0, u'outside_temp_min': 0, u'channels': 0, u'single': 0, u'duration': -1, u'samplerate': u'', u'system_tempf': 117.9, u'random_onoff': u'Off', u'system_tempc': 47.7, u'album': u [... cut ...]
  File "pydPiper.py", line 762, in <module>
    dc.load(pagefile, mc.musicdata,mc.musicdata_prev )
  File "/app/displays/display.py", line 1313, in load
    newpages = imp.load_source('pages', file)
  File "pages_ssd1306.py", line 47
    'ipaddr': { 'type':'text', 'format':'{0}', 'variables':['ip'], 'font':'large', 'varwidth':True },
           ^
SyntaxError: invalid syntax
`

The same if I add 'statictext': { 'type':'text', 'format':'this is my media player', 'font':'large', 'varwidth':True },'

Where am I wrong? 

thanks

manolomalele, 2020-05-13:
> You can create static text widgets the exact same way as widgets that use variables.
>
> `'statictext': { 'type':'text', 'format':'this is my media player', 'font':'large', 'varwidth':True },`
>
> You can [...]

Hi Dhrone, I've spent a couple of hours trying to figure it out, but I cannot understand the problem.
I've added the lines as you suggested. Then rebooted.
Nothing on the display.

Here it is the code:
pages_ssd1306.py
`#!/usr/bin/python.pydPiper
# coding: UTF-8

from __future__ import unicode_literals

# Page Definitions
# See Page Format.txt for instructions and examples on how to modify your display settings

# Load the fonts needed for this system
FONTS = {
        'small': { 'default':True, 'file':'latin1_5x8.fnt','size':(5,8) },
#       'large': { 'file':'BigFont_10x16.fnt', 'size':(10,16) },
        'large': { 'file':'Vintl01_10x16.fnt', 'size':(10,16) },
        'tiny': { 'file':'upperasciiwide_3x5_fixed.fnt', 'size':(5,5) },
}

TRUETYPE_FONTS = {
        'DejaVuSans28': { 'file':'DejaVuSans.ttf', 'size':28 }
}

IMAGES = {
        'progbar': {'file':'progressbar_80x8.png' },
        'splash': {'file':'pydPiper_fixed_splash.png' }
}

# Load the Widgets that will be used to produce the display pages
WIDGETS = {
        'xofy': { 'type':'text', 'format':'{0}/{1}', 'variables':['playlist_position', 'playlist_length'], 'font':'small', 'size':(58,16), 'varwidth':True },
        'volume': { 'type':'text', 'format':'Volume: {0}', 'variables':['volume'], 'font':'small', 'just':'right', 'size':(60,16), 'varwidth':True },
        'volumelarge': { 'type':'text', 'format':'Volume: {0}', 'variables':['volume'], 'font':'large', 'just':'left', 'varwidth':True },
        'volumebar': { 'type':'progressbar', 'value':'volume', 'rangeval':(0,100), 'size':(115,8) },
        'artist': { 'type':'text', 'format':'{0}', 'variables':['artist'], 'font':'large','varwidth':True,'effect':('scroll','left',1,1,20,'onloop',3,125)},
        'title': { 'type':'text', 'format':'{0}', 'variables':['title'], 'font':'small','varwidth':True,'effect':('scroll','left',1,1,20,'onloop',3,125)},
        'samplerate': { 'type':'text', 'format':'{0}', 'variables':['samplerate'], 'font':'small', 'just':'center','varwidth':True},
        'bitdepth': { 'type':'text', 'format':'{0}', 'variables':['bitdepth'], 'font':'small', 'just':'center','varwidth':True},
        'elapsed': { 'type':'text', 'format':'{0}', 'variables':['elapsed|strftime+%-M:%S'], 'font':'small','size':(30,8), 'varwidth':True},
        'length': { 'type':'text', 'format':'{0}', 'variables':['length|strftime+%-M:%S'], 'font':'small','size':(30,8),'just':'right','varwidth':True},
        'songprogress': { 'type':'progressbar', 'value':'elapsed', 'rangeval':(0,'length'), 'size':(62,8) },
        'time': { 'type':'text', 'format':'{0}', 'variables':['utc|timezone+Europe/Rome|strftime+%-I:%M'], 'font':'large', 'just':'right', 'varwidth':True, '$
        'ttime': { 'type':'ttext', 'format':'{0}', 'variables':['utc|timezone+Europe/Rome|strftime+%-I:%M'], 'font':'DejaVuSans28', 'just':'right', 'varwidth$
        'ampm': { 'type':'text', 'format':'{0}', 'variables':['utc|timezone+Europe/Rome|strftime+%p'], 'font':'tiny', 'varwidth':True },
        'tempsmall': { 'type':'text', 'format':'{0}', 'variables':['outside_temp_formatted'], 'font':'small', 'just':'right', 'size':(24,16) },
        'temphilow': { 'type':'text', 'format':'H {0}\nL {1}', 'variables':['outside_temp_max|int', 'outside_temp_min|int'], 'font':'small', 'just':'right', $
        'temp': { 'type':'text', 'format':'{0}', 'variables':['outside_temp_formatted'], 'font':'large', 'just':'center', 'size':(80,16) },
        'weather': { 'type':'text', 'format':'{0}', 'variables':['outside_conditions|capitalize'], 'font':'large','varwidth':True, 'effect':('scroll','left',$
        'ipaddr': { 'type':'text', 'format':'{0}', 'variables':['ip'], 'font':'large', 'varwidth':True },
        'statictext': { 'type':'text', 'format':'High Definition Media Player', 'font':'small', 'varwidth':True },
}

# Assemble the widgets into canvases.  Only needed if you need to combine multiple widgets together so you can produce effects on them as a group.
CANVASES = {
        'playing': { 'widgets': [ ('artist',3,10), ('title',3,26), ('bitdepth',22,45), ('samplerate',78,45), ('elapsed',3,56), ('length',98,56), ('songprogre$
        'stoptime': { 'widgets': [ ('ttime',10,20), ('ipaddr',20,56) ], 'size':(128,64) },
 'stoptimetemp_popup': { 'widgets': [ ('ttime',3,0), ('tempsmall',100,0), ('weather',8,47), ('temphilow',100,48) ], 'size':(128,64) },
        'volume_changed': { 'widgets': [ ('volumelarge',8,0), ('volumebar',8,18) ], 'size':(128,64) }
}

# Place the canvases into sequences to display when their condition is met
[... 41 lines cut ...]
    dc.load(pagefile, mc.musicdata,mc.musicdata_prev )
  File "/app/displays/display.py", line 1313, in load
    newpages = imp.load_source('pages', file)
  File "pages_ssd1306.py", line 47
    'ipaddr': { 'type':'text', 'format':'{0}', 'variables':['ip'], 'font':'large', 'varwidth':True },
           ^
SyntaxError: invalid syntax
Player status at exception
{u'ip': u'192.168.1.7', u'random': 0, u'outside_temp_min': 0, u'channels': 0, u'single': 0, u'duration': -1, u'samplerate': u'', u'system_tempf': 117.9, u'random_onoff': u'Off', u'system_tempc': 47.7, u'album': u'', u'current_time': u'', u'system_temp': 47.7, u'bitdepth': u'', u'musicdatasource': u'', u'disk_availp': 76, u'title': u'', u' [... cut ...]
  File "pydPiper.py", line 762, in <module>
    dc.load(pagefile, mc.musicdata,mc.musicdata_prev )
  File "/app/displays/display.py", line 1313, in load
    newpages = imp.load_source('pages', file)
  File "pages_ssd1306.py", line 47
    'ipaddr': { 'type':'text', 'format':'{0}', 'variables':['ip'], 'font':'large', 'varwidth':True },
           ^
SyntaxError: invalid syntax
`

The same if I add 'statictext': { 'type':'text', 'format':'this is my media player', 'font':'large', 'varwidth':True },'

Where am I wrong? 

thanks

dhrone, 2020-05-14:
Could I suggest that you rename your version of pages_ssd1306.py to pages_ssd1306.txt and then attach it as a file to this thread?  Pasting it is damaging the content of the file.

manolomalele, 2020-05-15:
ok, sorry!
I'll do asap!

manolomalele, 2020-05-16:
Here it is!
This is the original working file 
[pages_ssd1306.txt](https://github.com/dhrone/pydPiper/files/4638165/pages_ssd1306.txt)

manolomalele, 2020-05-16:
And this is the one I've modified according to your suggestion.
Doing this nothing is displayed, just a black screen.

After adding/removing a widget is it necessary to do sopmething else? like stop&restart the service? compile something?

Thank you for your time 

[pages_ssd1306.txt](https://github.com/dhrone/pydPiper/files/4638175/pages_ssd1306.txt)

manolomalele, 2020-05-16:
Uh, and i'ts the same (nothing on screen) even creating a new canvas with only one of the widgets you've suggested me.

dhrone, 2020-05-16:
You have a couple of minor errors in your definition of WIDGETS.  Basically each element of an array needs to have a comma at the end unless it is the last element in which case the comma is optional.  You also should be careful with tabs verses spaces.  

Here's a fixed version..
[pages_ssd1306 fixed.txt](https://github.com/dhrone/pydPiper/files/4638491/pages_ssd1306.fixed.txt)

manolomalele, 2020-05-16:
Hi Dhrone, nothing to do.
I've just cpoied/pasted the file you gave me and the screen doesn't start.
Here it is the error code.

pi@moode:~/pydPiper $ sudo  /usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/pi/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py
pydPiper starting
Page file pages_ssd1306.py was unable to be loaded
Uncaught exception
Traceback (most recent call last):
  File "/app/pydPiper.py", line 762, in <module>
    dc.load(pagefile, mc.musicdata,mc.musicdata_prev )
  File "/app/displays/display.py", line 1313, in load
    newpages = imp.load_source('pages', file)
  File "pages_ssd1306.py", line 47
    'statictext': { 'type':'text', 'format':'this is my media player', 'font':'small', 'varwidth':True },
               ^
SyntaxError: invalid syntax
Player status at exception
{u'ip': u'192.168.1.3', u'random': 0, u'outside_temp_min': 0, u'channels': 0, u'single': 0, u'duration': -1, u'samplerate': u'', u'system_tempf': 105.6, u'random_onoff': u'Off', u'system_tempc': 40.9, u'album': u'', u'current_time': u'', u'system_temp': 40.9, u'bitdepth': u'', u'musicdatasource': u'', u'disk_availp': 76, u'title': u'', u'playlist_count': -1, u'bitrate': u'', u'current_tempc': 40.9, u'current': -1, u'outside_temp_max': 0, u'state': u'starting', u'time_formatted': u'', u'type': u'', u'remaining': u'', u'localtime': <Moment(2020-05-17T00:00:26)>, u'actPlayer': u'', u'repeat': 0, u'elapsed': -1, u'volume': -1, u'playlist_display': u'', u'disk_avail': 11351470080L, u'playlist_length': -1, u'outside_conditions': u'No data', u'single_onoff': u'Off', u'system_temp_formatted': u'40\xb0c', u'outside_temp_formatted': u'', u'elapsed_formatted': u'', u'utc': <Moment(2020-05-16T22:00:26)>, u'disk_usedp': 24, u'repeat_onoff': u'Off', u'artist': u'', u'disk_used': 3449901056L, u'current_time_sec': u'', u'uri': u'', u'current_ip': u'192.168.1.3', u'playlist_position': -1, u'length': -1, u'current_tempf': 105.6, u'position': u'', u'current_time_formatted': u'', u'tracktype': u''}
Traceback (most recent call last):
  File "/app/pydPiper.py", line 762, in <module>
    dc.load(pagefile, mc.musicdata,mc.musicdata_prev )
  File "/app/displays/display.py", line 1313, in load
Loading font from /app/displays/fonts/latin1_5x8_fixed.fnt
    newpages = imp.load_source('pages', file)
  File "pages_ssd1306.py", line 47
    'statictext': { 'type':'text', 'format':'this is my media player', 'font':'small', 'varwidth':True },
               ^
SyntaxError: invalid syntax

I really don't understand...