dhrone/pydPiper #87: genre not a valid variable

NicoOosterwijk, 2020-02-13:
Superb piece of program, I love it and use it every day to display song infor on my 16x2 LCD display. However, I'm missing the music's genre.

Is it possible to add the music's genre as a variable so it an be used on the display?

tia,

.Nico

dhrone, 2020-02-15:
Currently no.  pydPiper is only set up to make the underlying distributions metadata available.  As far as I know none of the popular distributions (e.g. Volumio, Rune) provide genre data.  It would theoretically be possible to add a lookup function that pulled genre data from somewhere else but the system does not currently implement that functionality.

NicoOosterwijk, 2020-02-15:
Ok, I get it. Thanks for your prompt answer, please close this issue.

Rgds,

Nico

closed 2020-02-15 by dhrone