dhrone/pydPiper #88: 16x2 text getting pushed to the right by random char/empty spaces appearing on left of screen.

TaliesinDS, 2020-02-20:
My hd44780 lcd works otherwise perfectly but while playing every few hours or so the canvas? shifts 2 blocks to the right and it keeps happening and after hours/days it will move all the way to the right and the space on the left will be a mix of random? chars and empty space.

I usually have volumio playing just 1 webradio station.
Switching to a different station or something totally different like mp3 songs does not affect the screen but when i power the raspberry down the "exiting..." text appears on the correct position in the top left and all random chars and empty spaces disappear.

After a reboot everything works fine again until slowly the random stuff pushed the test off the screen to the right again. 

pic of the screen: https://drive.google.com/open?id=1fStIRT2dSe-wkvUoeuJEEzKHbrKe79nS

My system details: raspberry pi 4 with hifiberry dac+. adafruit hd44780, official rpi 4 powersupply.
volumio 2.699

dhrone, 2020-02-21:
These type of errors generally relate to timing issues with the display.  You can try adjusting the pulse timing setting to something higher to see if that has any impact.  If it doesn't, I would suggest trying a different HD44780 display.

TaliesinDS, 2020-02-21:
If i understand correctly that would be "self.delayMicroseconds(1) # 1 microsecond pause - enable pulse must be > 450ns" on line 330 in hd44780.py ? (i am using 4bit mode)

And how does it work with docker ? if i increase the delay, save and reboot it works in docker ?

I'm sorry for being such a noob lol, this is the first time i'm messing with linux.

dhrone, 2020-02-21:
Yes.  That's correct.  Normally that value would be adjustable in the pydPiper.cfg file but I saw that in the current version of the HD44780.py driver that it is not reading from the config file to determine what the pulse time should be.  You should try a longer value like 10 instead of the current value of 1.

Note:  Make sure that your ExecStart line includes the -v clause needed to reference the local copy of pydPiper, not the one that is inside the docker image itself.  It should look like this...

`ExecStart = /usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py
`

TaliesinDS, 2020-02-22:
So far it seems to be working, thanks for the help :)

closed 2020-02-22 by TaliesinDS