dhrone/pydPiper #89: Moode Audio

devGoodCat, 2020-02-21:
Hi !
Sorry for google translator.
First of all, thanks for your code!

I have a problem, I started installing your software for Modee Audio. 
It is written in your description that it is supported, but in the initialization line <ExecStart>, for the <musicdistro> parameter, I do not understand what needs to be specified.

I also tried to install according to the instructions:
http://moodeaudio.org/forum/showthread.php?tid=461
- the test was successful, but the page did not change when playing music (it showed the clock).

dhrone, 2020-02-21:
If the clock was displayed you are close.  Most of the system would need to be set up correctly to get to that point.  I suspect that there is something going on with your configuration of the source section of pydPiper.cfg.  

Look in that file and see what the [SOURCE] section says.  For a normal Moode installation it should look like this...

`[SOURCE]

source_type = moode
mpd_server = localhost
mpd_port = 6600
`

devGoodCat, 2020-02-21:
I assigned mpd_server = moode.local
it seems there was a mistake. But after I fixed, an error appeared: no active widget .
I read earlier how to fix it, I’ll leave it here, maybe someone will need it:
Replace any visible db['stream'] with db['encoding']
https://github.com/dhrone/pydPiper/issues/20#issuecomment-399179697

Best regards thanks more!

closed 2020-02-21 by devGoodCat