dhrone/pydPiper #90: How to add new system/status variable(s) to the PAGES file and service

mariuszmysz, 2020-02-25:
Hi dhrone,

I built a player based on RPI 2B and volumio.
Energy is managed by a Pico UPS with a lithium-ion battery. I would like to track the condition of this battery using your service.
Where should I start to add another variable to the service and then refer to it in the PAGES file?
Just like you handle the IP address or processor temperature. Is this even possible?
I'm not a python programmer, but Delphi / SQL - maybe I can do it?

marius

dhrone, 2020-03-03:
I would add it to the updatesystemvars method in pydPiper.py.  Perform whatever call you need to pull the state of the pico and then add the value within the with self.musicdata_lock: section.

 `               self.musicdata[u'picopower'] = pico_power`

Keep in mind that currently, updatesystemvars goes to sleep for 300 seconds after every execution.  If you need quicker updating, adjust the sleep value at the end of the method.

Note:  The way that the system variables is updated is somewhat awkward.  I will likely update it at some point to be more consistent with the way the different musicdata modules are structured.

TheHighFlyingBirds, 2022-02-10:
As this is still open, can the page files be edited to either run a command directly as a widget or have a widget that runs a script as opposed to having to modify the main pydPiper.py file. Just thinking for simplicity.  Only looking at a command like below, or I could wrap it as a script.

`top -bn1 | grep load | awk '{printf "CPU : %.0f\n", $(NF-2)}' `

Cheers

dhrone, 2022-02-10:
In the new version (unreleased) of pydPiper the answer would be yes.  But in the current version this is not possible.  The page files in the current version are read in once at start-up.  To get what you would want, the code would need to be re-executed periodically to update the value.  That feature is included in the new version.  

I can't commit to when that would be released.  If there is a particular distribution that would be desirable I can take that into account.  The distribution does need to be able to either host docker, or python 3.

TheHighFlyingBirds, 2022-02-12:
Sounds interesting, will keep an eye out for the new version. 

No particular distribution at the moment (ideally picoreplayer but understand thats a whole heap more work), as just trying volumio and moode at the moment to see if they give me what I want. Max2player would be an option but cant get touch support on the pi4 without a subscription / license.