dhrone/pydPiper #92: bitrate not working

NicoOosterwijk, 2020-03-21:
When trying to display the bitrate, the variable 'bitrate' does not show anything on the display.
Using volumio, on the cli with command 'volumio status' I can see the bitrate just fine with a value of 128 kbps.

dhrone, 2020-04-19:
Ok.  I just looked into this and found out that Volumio has only recently added bitrate to the list of status variables it supports.  I made a quick but untested change to musicdata_volumio2.py to address this.  If you are running the latest code from the 'master' branch just pull the latest and give it a try.  

If you are not running from the 'master' branch, let me know what specific version you are running and how it was installed and I'll try to give you direction to install the change.

manolomalele, 2020-04-21:
Hi, Is it also possible to add a widget for the file type and the current ip? 

Thank you

NicoOosterwijk, 2020-04-23:
perfect! Works like a charm, thank you!

closed 2020-04-23 by NicoOosterwijk