dhrone/pydPiper #93: Support for openhome players?

Dual01, 2020-04-13:
Dear dhrone,
pydPiper is a great program, many thanks and congrats to you dhrone! I switched from an mpd-based player to the ohplayer from openhome.org. PydPiper supports several players such as volumio (v1 and v2), MoodeAudio, and other mpd-based players but cannot retrieve the status and the metadata from an openhome player. It would be great if PydPiper would have also the option to work in an openhome system. There is a wiki how to retrieve  information like player status, metadata etc. on openhome.org but my python knowledge is to low in order to be able to implement this. Thus, my question: would you be willing to include such an option?

dhrone, 2020-04-19:
I can look into it.  Please provide a direct link to the documentation that explains how to retrieve the player information.

Dual01, 2020-04-19:
There is a description for addressing openhome devices with python:
https://pypi.org/project/openhomedevice/
and here is the documentation on the openhome.org site:
http://wiki.openhome.org/wiki/OhMediaDevelopers

dhrone, 2020-04-21:
Thanks.  It seems reasonably clear-cut but would require the development of a new music data handler for the system.  I'm looking into refactoring that part of the code so will see if I can add support when I pick that up.  Unfortunately, I cannot give you an estimate for when I will have that completed.

Dual01, 2020-04-21:
Thanks for checking