dhrone/pydPiper #94: Volumio2 not working

fabio323ti, 2020-04-25:
reinstalled many times, always same error.
i also copied pydpiper.service into /etc/systemd/system, enabled and started.. no way

if i load manually, got this error:

(comand i'd used : sudo /usr/bin/docker run --network=host --privileged -ti -v /var/log:/var/log:rw -v /home/volumio/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha /bin/bash python pydPiper.py )

`root@volumio:/app# python pydpiper.py
python: can't open file 'pydpiper.py': [Errno 2] No such file or directory
root@volumio:/app# python pydPiper.py
Traceback (most recent call last):
  File "pydPiper.py", line 23, in <module>
    class music_controller(threading.Thread):
  File "pydPiper.py", line 68, in music_controller
    'localtime':moment.utcnow().timezone(pydPiper_config.TIMEZONE),
  File "/usr/local/lib/python2.7/dist-packages/moment/core.py", line 81, in timezone
    date = times.to_local(date, zone)
  File "/usr/local/lib/python2.7/dist-packages/times/__init__.py", line 61, in to_local
    'First argument to to_local() should be a universal time.'
ValueError: First argument to to_local() should be a universal time.
`

dhrone, 2020-05-09:
What version of Volumio are you on and what TIMEZONE do you have configured in pydPiper.cfg?

mdphoto, 2021-01-24:
Hi,
I have the same log, i just set the time zone in Volumio with sudo dpkg-reconfigure tzdata

Saiyato, 2022-03-03:
I'm actually seeing the same problem in my attempt to build a new (and very small) container. As it appears this error seems to be caused by a None-value for pydPiper_config.TIMEZONE. 

Maybe that it helps someone, I actually didn't copy the pydPiper.cfg file into the container, just for testing purposes (checking if it's a necessary part). :wink: So check if this file is actually there, and if the timezone value makes sense.

paudio25, 2022-12-29:
I'm using Volumio 3.396 and ran into the same problem. In my case was the solution in pydpiper.service: the --timezone value must match with dpkg-reconfigure tzdata. Mine is 'Europe/Amsterdam'. I hope this can help.