dhrone/pydPiper #98: How to configure for 6 pin serial display?

ashrond, 2020-05-17:
I have gotten Volumio 2 installed on a raspi3b and followed the process for deploying a docker container and then installing PydPiper.. the issue i am running into is i have a 6 pin serial display. I have been reading through the documentation but i cannot seem to find the proper driver /config for this. as when running through the install.sh it returns 
Possible values are: winstar_weg, hd44780, hd44780_i2c, hd44780_mcp23008, luma_i2c

I have gotten the display to output messages from a python script in the past, but cannot for the life of me find the document that outlines working with a 6pin serial display, 

any guidance you can provide would be greatly appreciated.

dhrone, 2020-05-18:
Please provide the specific make and model of display you are trying to use

ashrond, 2020-05-18:
Noritake-ITRON CU16024-UX3J
https://noritake-vfd.com/cu16024-ux3j.aspx

It says it should emulate the HD44780, however the config for pydpiper is asking for a pin config that wont work with this board since it is serial (that is my understanding anyway) 

I have been able to successfully test-write data to the screen using the pin config and process outlined here: https://projects-raspberry.com/interfacing-a-vfd-display-to-the-raspberry-pi/

dhrone, 2020-05-18:
Thanks.  The current display interface types that are supported are parallel and i2c.  Your display is using SPI.  It doesn't look terribly difficult to implement the support for it but I currently lack a display that uses SPI for testing.

ashrond, 2020-05-18:
if you provide a contact address so its not posted in a public form, i am happy to send you this display if it means getting support for SPI implemented.  I was thinking of upgrading to the 4 line display for this project anyway.

ashrond, 2020-05-20:
I have a new 4 line display version on its way to me, let me know if you want me to send you this one for testing. if not then perhaps i can facilitate testing somehow?

ashrond, 2020-06-05:
> Thanks. The current display interface types that are supported are parallel and i2c. Your display is using SPI. It doesn't look terribly difficult to implement the support for it but I currently lack a display that uses SPI for testing.

just following up, i have a screen i can send you for Rnd/testing. 
if you want to ping me spam.th@outlook.com

dhrone, 2020-11-11:
I'm finally at a point where I can start looking at new display types.  So, it is possible I could implement a driver for this display.  That said, as noted previously, this display does support a parallel-bus style interface.  Any reason, not to give that a shot using the hd44780 driver within pydPiper?

FYI, I'm currently working on a re-write of pydPiper that moves the code-base to Python3.  Part of that effort was to integrate all of the display logic into the Luma.Oled (and Lume.Lcd) projects.  Luma does support SPI natively.  If you can get your display running using that library it would be much simpler for me to get a patch enabled in the current version of pydPiper to support you.