dhrone/pydPiper #99: WeatherUnderground API shutdown

ghiglie, 2020-05-29:
Hi,
 I'm getting some 503 errors from WU APIs in log:
`The Weather Underground server couldn't fulfill the request and responded with error code 503
The Weather Underground server couldn't fulfill the request and responded with error code 503
`

and I noticed [it's been shut down](https://apicommunity.wunderground.com/weatherapi/topics/end-of-service-for-the-weather-underground-api).

Is it being dropped in pydPiper too?

dhrone, 2020-06-23:
Yup.  I'm working on a capability to pull the data from RSS feeds.  This would allow lots of new services to be supported many of which do not require registration such as the BBC Weather RSS feed.  See https://www.bbc.com/weather/about/17543675

ghiglie, 2020-06-23:
...Really good news! Thanks for your time.

closed 2020-06-23 by ghiglie