dhrone/pydPiper #102: SSD1306 Font Problem

rh1017, 2020-06-19:
Hi,
I am using an SSD1306 128x64 OLED display with moOdeaudio 6.5.2, Pi3B+ with Raspian Buster. Text strings using both small and large fonts display apostrophe ' as ? question marks. All other characters display correctly.
Is this a known issue and is there a fix?
Thanks,
Richard.

dhrone, 2020-06-23:
Not that I am aware of.  Can you let me know specifically what fonts (by filename) you are using that are displaying the apostrophe as a question mark?  If you are using one of the pre-canned pageFiles, let me know which one and you can just let me know what fields are displaying the question marks.  If you are using a modified pageFile, please provide it.

rh1017, 2020-06-24:
Thanks for your reply. The plot deepens....!
I thought that a picture might have been of assist and much to my surprise (and potential embarrassment)...
[image]
- both apostrophe and question mark reproduce as they should.
So, bacto the original file that triggered the post ...
[image]
and here is the way the track is displayed on the moOde GUI in Windows Chrome browser...
[image]
Anumber of other files tested display correctly and there are also some that don't
[image]
..

dhrone, 2020-06-24:
Ok, I think I understand what the problem is.  It appears that in some of the metadata for the songs you are playing, they are using the unicode value for 'right single quotation mark' (u+2019) instead of a simple apostrophe (u+0027).  The Vintl01_10x16_fixed.fnt does not contained that character.  There is a simple fix though.  You can add another line to the font file that uses the same data as apostrophe.  Try this file as a replacement for your Vintl01_10x16_fixed.fnt file and see if the problem goes away.  Keep in mind there are other characters that could also trip you up like 'left single quotation mark'.  Solving for these can be handled similarly or since you are using a graphical display (e.g. not a HD44780), you could use a truetype font that contains a broader symbol set.  It would also be possible to add new symbols to the Vint01_10x16 but that is a more involved process.

[Vintl01_10x16_fixed_test.fnt.txt](https://github.com/dhrone/pydPiper/files/4826711/Vintl01_10x16_fixed_test.fnt.txt)

rh1017, 2020-06-25:
Unfortunately that did not work. However I edited the metadata for artist and track and simply replaced the meta apostrophe with the Windows apostrophe and both title and track display correctly in pydPiper with the packaged pages_ssd1306.py and font files.
I'm not familiar with the workings of fonts and character mapping but for the relatively few (hopefully) files that display incorrectly, the metadata edit is an easy fix.
Thanks very much for helping with this question and if there is anything else I can try I would still like to find out why.
Regards, 
Richard.

dhrone, 2020-11-11:
I'm going to close this since it is related to the font itself and not the system.  Better font handling may be available in a future release.

closed 2020-11-11 by dhrone