dhrone/pydPiper #103: Problem with pydPiper and 20 x 4 lcd

cockneynutjob, 2020-07-15:
Hello, im extremely confused! Ok ive got the latest version of volumio and i want to add a 20 x 4 lcd, ive tried everything and still cant get it to work been 3 days ive tried every way possible, your guide on here does not work either for me, i see other people useing it but i cant get it to work, when i do  sudo ./install.sh at the end all i get is bad message missing =? The display never starts 
Please could you help me because a raspberry pi is going to be in little pieces soon lol thanks

dhrone, 2020-07-17:
I'll need some additional details.
1.  What specific display are you using and what type of interface does it have?
2.  Have you gotten any other programs to display test on the display?  Assuming you are using an HD44780 compatible LCD you might check [(https://tutorials-raspberrypi.com/raspberry-pi-lcd-display-16x2-characters-display-hd44780/)](url)
3. What version of pydPiper are you using?  Did you pull from the current main branch `git clone https://github.com/dhrone/pydPiper` or did you get it from somewhere else?
4.  What did you do (step-by-step) to get the error message?
5.  What is the full text of the error message?

cockneynutjob, 2020-07-17:
Dont worry i have sorted it now i just used the pydpiper plugin, thanks for the reply appreciate it, couldn't get the way your github page said to do it but the plugin worked pretty much straight away, just some constructive criticism both this guide and his plugin guide are extremely vague, i had to go though about 50 different sites to get this to work, there are loads of dependences that need to be installed before these can work, i might do a youtube video on how to set this up as theres alot of people having problems with it, but the display looks great now so thank you, and for the reply again.

dhrone, 2020-07-22:
Sharing your experience would be valuable for the community.  Thanks!

closed 2020-07-22 by dhrone