dhrone/pydPiper #104: HD44780_i2c blue LCD not properly working with on MAX2PLAY connected to seperate LMS

Klaasemann, 2020-07-22:
Hello Dhrone,

i installed latest pydPiper on latest MAx2play. Display is connected to 0x27 on port 1. LMS is connected, confirmed by log file.

After "pydPiper Starting..." the nice clock appears and thats all. When shutting down the message "Exiting.." appears.
No Information of song or other infos are displayed.  

Please see last entry of log:
`2020-07-22 12:49:37,897:INFO:Exiting...
2020-07-22 12:51:41,414:INFO:pydPiper starting
2020-07-22 12:51:41,559:DEBUG:Loading display controller
2020-07-22 12:51:41,559:DEBUG:Loading music controller
2020-07-22 12:51:41,573:DEBUG:LMS musicdata service starting
2020-07-22 12:51:41,573:DEBUG:Connecting to LMS service on 192.168.0.101:9090
2020-07-22 12:51:42,257:DEBUG:Connected to LMS using player YamahaPi
2020-07-22 12:51:42,391:DEBUG:Connecting to LMS raw service on 192.168.0.101:9090
2020-07-22 12:51:43,576:DEBUG:Music Controller Starting
2020-07-22 12:51:43,577:DEBUG:Loading pages_lcd_20x4.py as page file
2020-07-22 12:51:43,579:DEBUG:Initializing current system status update process.  System status will update every five minutes
2020-07-22 12:51:43,621:DEBUG:Weather service not enabled
2020-07-22 12:51:43,674:DEBUG:Weather service not enabled
2020-07-22 12:51:43,714:DEBUG:System status: Temp 53Â°c, Disk space remaining 74%, IP address 192.168.0.202
2020-07-22 12:51:43,778:DEBUG:Loading image progbar
**2020-07-22 12:51:43,806:DEBUG:v out of range with value -1.  Should have been between 0 and 100**
`

Hopefully you can point where my fault is..

Thanks in advance
Klaas

dhrone, 2020-07-22:
It looks like the LMS data service is successfully connecting.  I'd have to download the current Max2play to see if there has been some change to the way it is transmitting LMS data.  Can you confirm the version of Max2Play and the version of LMS you are running?

Klaasemann, 2020-07-22:
I installed 249 for PI 3, not Pi4.

https://www.max2play.com/max2play-image/

 

LMS Version 8 - 1594451286 @ Sat Jul 11 09:25:03 CEST 2020

·         Betriebssystem: piCore - DE - utf8

·         Plattformarchitektur: armv8l-linux

·         Perl-Version: 5.28.2 - arm-linux-gnueabihf-thread-multi-64int

·         Audio::Scan: 1.02

·         IO::Socket::SSL: 2.066

·         Datenbankversion: DBD::SQLite 1.58 (sqlite 3.22.0)

 

 

 

Regards

Klaas

Klaasemann, 2020-07-27:
Hello,

last weekend i just created a new fresh installation based on latest version (249).
Then followed the instructions. Now its working. The Titles and infos are shown.
But:
After some title changes and using snchronisation, i just see the clock.
This is the main player and after a reboot it works again..

Do you have an idea ?  

Regards
Klaas

dhrone, 2020-07-28:
Usually this type of behavior occurs when their is a problem receiving metadata from the LMS server.  The time issue is probably unrelated.

Klaasemann, 2020-08-19:
Hello,

i tried the installation prezess several times on Raspi 2 and 3b+ with hd44780 20x4 on max2play 249.
pydPiper works properly just on devices where LMS is locally installed on. Showing clock and showing text of playing tracks.
pydPiper on a LMS client does just show the clock. No text.

I just got following errorlog on ALL clients:
----------------------------------------------------------------------------------------------------
2020-08-19 17:22:16,595:ERROR:LMS connect failure
Traceback (most recent call last):
  File "/app/sources/musicdata_lms.py", line 96, in connect
    self.dataserver.connect()
  File "/usr/local/lib/python2.7/dist-packages/pylms/server.py", line 58, in connect
    self.telnet_connect()
  File "/usr/local/lib/python2.7/dist-packages/pylms/server.py", line 66, in telnet_connect
    self.telnet = telnetlib.Telnet(self.hostname, self.port)
  File "/usr/lib/python2.7/telnetlib.py", line 209, in __init__
    self.open(host, port, timeout)
  File "/usr/lib/python2.7/telnetlib.py", line 225, in open
    self.sock = socket.create_connection((host, port), timeout)
  File "/usr/lib/python2.7/socket.py", line 571, in create_connection
    raise err
----------------------------------------------------------------------------------------------------

Any idea hint ?

Regards
Klaas

dhrone, 2020-09-14:
The problem is in the pylms library but its not clear why the connection request is failing.  Unfortunately, I have not had time to load the current max2play client to see what may have changed.

Klaasemann, 2020-09-14:
Can you tell me on wich distro it is working ?`

 

Thanks in adcance

dhrone, 2020-11-11:
This appears to be the same issue as #109.  Are you attempting to connect to an LMS service located on a different server?

TheHighFlyingBirds, 2022-02-18:
Hi Dhrone,

Sorry for all my recent questions. I have been trying to get this to run on max2play to see if that will serve my needs, but seem to be having an issue with getting anything to display at all. Running Max2play (version 2.55) on pi4 2gb, and have LMS on a second pi (IP 192.168.0.10).

I have cloned pydpiper from main branch, and run the install docker script, and then ran the configure.py script, and copied the service file to the systemd/system/ folder and enable and start pydpiper.  Both pydpiper and docker are showing as running using sudo systemctl status. Attached are my service and cfg files. 

[pydpiper.service.txt](https://github.com/dhrone/pydPiper/files/8096583/pydpiper.service.txt)
[pydPiper.cfg.txt](https://github.com/dhrone/pydPiper/files/8096472/pydPiper.cfg.txt)

For the service file, I have tried it with both options enabled for the exec start (and ensured copied to /etc/systemd/system/). I note in the service file, in the [Unit] section, it states 'After = volumio.service' I assume this should be changed as I am not running volumio, although it didn't affect anything when I ran pydpiper on Moode. 

When I run the following `sudo /usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/pi/pydPiper:/app:rw dhrone/pydpiper:latest python /app/pydPiper.py` I get the following error:

python /app/pydPiper.py
pydPiper starting
Uncaught exception
Traceback (most recent call last):
  File "/app/pydPiper.py", line 739, in <module>
    lcd = displays.hd44780_i2c.hd44780_i2c(rows, cols, i2c_address, i2c_port, enable)
  File "/app/displays/hd44780_i2c.py", line 114, in __init__
    self.bus = smbus.SMBus(i2c_bus)
IOError: [Errno 2] No such file or directory
Traceback (most recent call last):
  File "/app/pydPiper.py", line 739, in <module>
GPIO not installed
    lcd = displays.hd44780_i2c.hd44780_i2c(rows, cols, i2c_address, i2c_port, enable)
  File "/app/displays/hd44780_i2c.py", line 114, in __init__
    self.bus = smbus.SMBus(i2c_bus)
IOError: [Errno 2] No such file or directory

After reading issue #106 I ran `sudo docker container ls -a` and that just shows a large list of containers which are created but keep exiting. Assuming it is to do with max2play, as it works beautifully on volumio using same install methods. 

Thanks for all your effort and apologies for my many questions.