dhrone/pydPiper #105: display offset

funhouse61, 2020-07-30:
Hi !
First of all thank you for the job done !
 my raspi is b+ v1 2 and oled display midaseh002004a 20x4 with moode audio 6.2.
 
  My first installations were going normally, but since recently when I do the test everything is ok and after the final installation I have this kind of display
[image]
and it continues even when I start the music, yet I haven't changed anything in the way I install pydPiper.
otherwise I am doing tests on this model of raspberry to familiarize myself with pydpiper and the modification of my page eventually the screen will be coupled with a raspi 3 and a digione card and a sofstart by audiophonics with gpio replicator.
and on this subject should I modify anything, wiring or the script of pydPiper so that everything works together ?
sorry I still have questions, I find that the refresh rate is very long the display is always half a song behind the reality is it due to the raspi which is not the most efficient or I can change something? my screen is wired in parallel to the fact.
last question I tried to integrate tracktype on the screen but once installed here is what it gives.
[image]
On the other hand, if you need my pydpiper configuration, you should also explain to me how to get them back to you and send them to you.
sorry for all these questions but i'm starting and i don't know much about it.
I specify that I searched the various forums and on github before posting here but that I did not find my answers.
Please help me !!!

funhouse61, 2020-07-31:
hello, so try with the latest version of moode freshly downloaded.
So, as in the Remy1961 instruction guide on the moodeaudio forum I do, the test.
That's what happens.
[image]
[image]
and after final installation.
[image]
I try to restart pydpiper,
[image]
It's worse.
I will start all over but without doing the test and I will see.

dhrone, 2020-08-01:
Hmm.  So it works when you issue the following command...

`sudo /usr/bin/docker run --network=host --privileged -ti -v /var/log:/var/log:rw -v /home/pi/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha /bin/bash
python pydPiper.py`

When does it not work and what specific command is being used to start it?  If you are starting it from systemd let me know what the contents of pydpiper.service are.  Also let me know what the contents of pydPiper.cfg are.  

My guess is that the settings for the display pins are incorrect.

funhouse61, 2020-08-01:
Hello,
  so after the test I type this:
cd pydPiper
sudo cp pydpiper.service /etc/systemd/system
sudo systemctl enable pydpiper
sudo systemctl start pydpiper

funhouse61, 2020-08-01:
now I will try to send you the data you request

funhouse61, 2020-08-01:
[Unit]
Description = pydPiper
Requires = docker.service
After = mpd.service docker.service

[Service]
Restart = always
ExecStart = /usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/pi/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py

[Install]
WantedBy = multi-user.target

funhouse61, 2020-08-01:
import logging
import sys
if sys.version_info[0] < 3:
    import ConfigParser
    config = ConfigParser.RawConfigParser()
else:
    import configparser
    config = configparser.RawConfigParser()

config.read('pydPiper.cfg')

def safeget(config, section, option, default=None):
    return config.has_option(section, option) and config.get(section, option) or default

# Start-up mode
STARTUP_MSG_DURATION = float(safeget(config,'STARTUP', 'startup_msg_duration',0))

# Display Parameters
DISPLAY_DRIVER= safeget(config,'DISPLAY', 'display_driver')
DISPLAY_DEVICETYPE= safeget(config,'DISPLAY', 'display_devicetype')
DISPLAY_WIDTH = int(safeget(config,'DISPLAY', 'display_width',0)) # the  width of the display in pixels
DISPLAY_HEIGHT = int(safeget(config,'DISPLAY', 'display_height',0)) # the height of the display in pixels
DISPLAY_SIZE = (DISPLAY_WIDTH, DISPLAY_HEIGHT)
DISPLAY_PIN_RS = int(safeget(config,'DISPLAY', 'display_pin_rs',0))
DISPLAY_PIN_E = int(safeget(config,'DISPLAY', 'display_pin_e',0))
DISPLAY_PIN_D4 = int(safeget(config,'DISPLAY', 'display_pin_d4',0))
DISPLAY_PIN_D5 = int(safeget(config,'DISPLAY', 'display_pin_d5',0))
DISPLAY_PIN_D6 = int(safeget(config,'DISPLAY', 'display_pin_d6',0))
DISPLAY_PIN_D7 = int(safeget(config,'DISPLAY', 'display_pin_d7',0))
DISPLAY_PINS_DATA = [ DISPLAY_PIN_D4, DISPLAY_PIN_D5, DISPLAY_PIN_D6, DISPLAY_PIN_D7 ]
i2c_address = safeget(config,'DISPLAY', 'display_i2c_address','0')
DISPLAY_I2C_ADDRESS = int(i2c_address) if i2c_address and 'x' not in i2c_address else int(i2c_address,16)
DISPLAY_I2C_PORT = int(safeget(config,'DISPLAY', 'display_i2c_port',0))
DISPLAY_ENABLE_DURATION = float(safeget(config,'DISPLAY', 'display_enable_duration',0)) # in microseconds.  Decrease to increase performance.  Increase to improve display stability

# Page Parameters
PAGEFILE = safeget(config, 'DISPLAY', 'pagefile')
ANIMATION_SMOOTHING = float(safeget(config,'DISPLAY', 'animation_smoothing',0)) # Amount of time in seconds to wait before repainting display

# System Parameters
# This is where the log file will be written
LOGFILE=safeget(config,'SYSTEM','logfile')

# Logging level
LOGLEVEL={'debug': logging.DEBUG, 'info': logging.INFO, 'warning': logging.WARNING, 'critical': logging.CRITICAL }.get(safeget(config,'SYSTEM', 'loglevel'))

# Localization Parameters
# Adjust this setting to localize the time display to your region
TIMEZONE=safeget(config,'SYSTEM', 'timezone')
TIME24HOUR=bool(safeget(config, 'SYSTEM', 'time24hour',False))

# Adjust this setting to localize temperature displays
TEMPERATURE=safeget(config,'SYSTEM', 'temperature')

# Weather related variables
WEATHER_SERVICE = safeget(config,'SYSTEM', 'weather_service')
WEATHER_API = safeget(config,'SYSTEM', 'weather_api')
WEATHER_LOCATION = safeget(config,'SYSTEM', 'weather_location')

# Music Source Parameters
MUSIC_SERVICE = safeget(config, 'SOURCE', 'source_type')

# Used by Volumio V1 and Moode
MPD_SERVER = safeget(config, 'SOURCE', 'mpd_server')
MPD_PORT = safeget(config, 'SOURCE', 'mpd_port')
MPD_PASSWORD = safeget(config, 'SOURCE', 'mpd_password')

# Used by Volumio v1 and Moode
SPOP_SERVER = safeget(config, 'SOURCE', 'spop_server')
SPOP_PORT = safeget(config, 'SOURCE', 'spop_port')
SPOP_PASSWORD = safeget(config, 'SOURCE', 'spop_password')

# Used by Volumio v2
VOLUMIO_SERVER = safeget(config, 'SOURCE', 'volumio_server')
VOLUMIO_PORT = safeget(config, 'SOURCE', 'volumio_port')

# Used by RuneAudio
RUNE_SERVER = safeget(config, 'SOURCE', 'rune_server')
RUNE_PORT = safeget(config, 'SOURCE', 'rune_port')
RUNE_PASSWORD = safeget(config, 'SOURCE', 'rune_password')

# Used by Max2Play and piCorePlayer
LMS_SERVER = safeget(config, 'SOURCE', 'lms_server')
LMS_PORT = safeget(config, 'SOURCE', 'lms_port')
LMS_USER = safeget(config, 'SOURCE', 'lms_user')
LMS_PASSWORD = safeget(config, 'SOURCE', 'lms_password')
LMS_PLAYER = safeget(config, 'SOURCE', 'lms_player')

I hope that's what you ask

funhouse61, 2020-08-01:
so here's how to pretend to be an idiot !!
I was forced, at the moment to start manually pydpiper because it does not start by itself and the display is organized correctly except that I am missing some information I put you a photo

funhouse61, 2020-08-01:
while waiting for it to arrive on my cloud I will explain; I am missing the name of the artist as in the basic configuration because I haven't changed anything yet

funhouse61, 2020-08-01:
I'm going to restart pydpiper to see if it doesn't start to display strangely again
and that's exactly what i was afraid of but just for the stop screen then for the moode info everything is ok. there I understand nothing more in fact it works randomly.

funhouse61, 2020-08-01:
at startup today
[image]
at restart
[image]
as if there is a big hole in the middle
and now with moode data it run normaly
[image]

funhouse61, 2020-08-01:
Finally I just noticed that the arrangement of the information is random anyway

funhouse61, 2020-08-01:
here are my changes in pages_lcd_20x4.py
[image]
and this is what is displayed. I am missing a part
[image]

dhrone, 2020-08-02:
You've sent the pydPiper_config.py file, not the pydPiper.cfg file which is the one I needed.  FYI, you can attach files to these messages.  It only supports a few file extensions though so you'll need to rename the file first.  In this case, create a copy of pydPiper.cfg and renaming it to pydPiper.cfg.txt.  You should then be able to attach it to your reply.  Also, can you provide a link to a description of your display.  I didn't find anything searching for midaseh002004a.

funhouse61, 2020-08-02:
display's informations
http://www.farnell.com/datasheets/2566408.pdf?_ga=2.58301994.1809785672.1593954716-188445626.1592058948&_gac=1.79231718.1593369053.CjwKCAjw_-D3BRBIEiwAjVMy7G56oSHiRySIxgz-L9ZUX19eiDEjSp0Xii87rE6iqiyeBslytlAlSRoCNnkQAvD_BwE
I'm sorry, but I can't make a copy of the file. i am trying with winscp i am not sure how.

funhouse61, 2020-08-02:
https://fr.farnell.com/midas/mcob42005a1v-egp/ecran-oled-20x4-parallele-5-3v/dp/2606856#

funhouse61, 2020-08-02:
[STARTUP]
startup_msg_duration = 5

[DISPLAY]
display_driver = hd44780
display_pin_rs = 7
display_pin_e = 8
display_pin_d4 = 25
display_pin_d5 = 24
display_pin_d6 = 23
display_pin_d7 = 27
display_width = 100
display_height = 32
display_enable_duration = 0.1
pagefile = pages_lcd_20x4.py
animation_smoothing = 0.15

[SYSTEM]
logfile = /var/log/pydPiper.log
loglevel = info
timezone = Europe/Paris
time24hour = true
temperature = celsius

[SOURCE]
source_type = moode
volumio_server = localhost
volumio_port = 3000
mpd_server = localhost
mpd_port = 6600
mpd_password = moodeaudio

funhouse61, 2020-08-02:
Is that right ?

funhouse61, 2020-08-02:
it gives the impression that it behaves like a 16x2

dhrone, 2020-08-04:
Hmm.  I cannot locate a manual that describes the controller that your display uses.  The manufacturer refers to it as an MS0010 but does not provide a manual on their site.  I'm curious though.  There is a controller made by winstar called the ms0010 that is used with their line of oled displays.  I  wonder if the MS0010 is a clone of it?  If that is the case, there may be small differences in the initialization routine between the hd44780 driver and one your display is using that is causing the problem.  Unfortunately, you cannot just use the winstar driver as it is a graphical driver and it is likely that your display only supports character mode.  I could see about adding a new driver that uses the initialization routine for the winstar but the rest of the driver is the hd44780 commands for you to test but it is a bit of a long shot.

funhouse61, 2020-08-04:
thank you already, for such extensive research. I don't mind waiting at all, I'm stuck anyway.
  It's bad luck for me, I just wanted a colored oled screen on a black background, which is not easy to find especially with drivers corresponding to your work or winstar or hd44780.
  but what strikes me as weird is that during the test it seemed to work correctly. I don't know what changes between the testing period and the finalization of remy1961 but could also be a lead. I say that, but hey as I don't know much about it so I may be wrong.
  Thanks again and I'll tell you in the next news

dhrone, 2020-08-04:
I strongly suspect a timing issue with the display.  One quick thing you could try is increasing the `display_enable_duration`.  You've currently got it set at `0.1`.  Try `0.2` or even `1.0`.

But, I think it will likely require a change to the initialization code to get it working reliably.

funhouse61, 2020-08-04:
Ok but could you explain me how can i change this.
With putty ? sorry i'm learning

funhouse61, 2020-08-04:
so i tried to modify with this command  "sudo python configure.py" but that didn't change anything.
otherwise I sent an email to midas asking them to help us. You never know it can work !?

dannixDanny, 2020-08-09:
the last config file you posted which is pydPiper.cfg has display_enable_duration = 0.1 about 10 lines down. Edit it to "0.2 or even 1.0" with a terminal editor.

funhouse61, 2020-08-10:
> le dernier fichier de configuration que vous avez posté qui est pydPiper.cfg a display_enable_duration = 0.1 environ 10 lignes plus bas. Modifiez-le en "0.2 ou même 1.0" avec un éditeur de terminal.

I have already tried this method but it hasn't changed anything.
thank you

dhrone, 2020-08-13:
I contacted the manufacturer to see if I could get a manual for the device but never heard back from them.

funhouse61, 2020-08-13:
so I guess you can't do anything without this manual?

funhouse61, 2020-08-27:
just to know if i should buy another display is there any chance you can help me without this manual?

dhrone, 2020-08-27:
Without a manual that includes the instruction set, no.

closed 2020-11-05 by dhrone