dhrone/pydPiper #106: Remote LMS server config help

dannixDanny, 2020-08-07:
Ive installed as per:

`sudo wget https://github.com/dhrone/pydPiper/archive/v0.3-alpha.tar.gz
sudo tar zxvf v0.3-alpha.tar.gz
cd pydPiper-0.3-alpha`

pydPiper.service says "can't assign requested address" but no mention of where to configure that address, is it an IP or a MAC?. Ive been poking about and updated pydPiper_config.py manually with the remote LMS ip and the MAC of the local squeezeplayer (MAC also on the command line) but have the same error.
`
pydPiper starting
LMS connect failure
Traceback (most recent call last):
  File "/app/sources/musicdata_lms.py", line 96, in connect
    self.dataserver.connect()
  File "/usr/local/lib/python2.7/dist-packages/pylms/server.py", line 58, in connect
    self.telnet_connect()
  File "/usr/local/lib/python2.7/dist-packages/pylms/server.py", line 66, in telnet_connect
    self.telnet = telnetlib.Telnet(self.hostname, self.port)
  File "/usr/lib/python2.7/telnetlib.py", line 209, in __init__
    self.open(host, port, timeout)
  File "/usr/lib/python2.7/telnetlib.py", line 225, in open
    self.sock = socket.create_connection((host, port), timeout)
  File "/usr/lib/python2.7/socket.py", line 571, in create_connection
    raise err
error: [Errno 99] Cannot assign requested address
^CTraceback (most recent call last):
  File "/app/pydPiper.py", line 627, in <module>
    time.sleep(2)
`

I prove network to LMS is ok using this:
` sudo /usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw dhrone/pydpiper:latest python /app/utils/lmslist.py -s 192.168.1.80`

it responds with the correct information.

`Workshop is at address b8:27:eb:40:10:be 
Bathroom is at address b8:27:eb:02:bc:e4  <---- player id like to monitor
`

I think I have a config issue but unsure how to correctly specify the IP for LMS as pydPiper_config.py having the IP didn't fix. I've been testing with this cmd

`sudo /usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw dhrone/pydpiper:latest python /app/pydPiper.py --lms --driver ssd1306_i2c --width 128 --height 64 --i2caddress 0x3c --i2cport 1 --timezone 'UK/London' --temperature celsius --lmsplayer 'b8:27:eb:40:10:be' --pages pages_ssd1306.py
`
Can anyone point me in the right direction?

dannixDanny, 2020-08-07:
I've just realised, it's running in Docker, so after some digging Ive blagged my way through docker, got a shell, edit the config and run pydPiper.py from the container shell. it didn't complain about LMS. Still need a method to set the IP of LMS that will stick but i cant edit the image. Regardless the oled was just jumping about and looked like a mirror image of junk, as in what was on the left of screen was repeated on the right.

dannixDanny, 2020-08-09:
Ive run docker with the -v option so i can edit files. I also downloaded the master code rather than the alpha stated in the install instructions. Then it was complaining about a missing font so installed that, still complained until I gave a full path to the file in pages, altho all the other fonts are relative??? so now I have the time, in the wrong zone it seems and thought It wasn't updating the music played. Switched logging to debug and its not monitoring the player I want, with the MAC i gave... 

Im not a docker expert obviously, I won't lie, i've found this install to be a nightmare. I'll plug on as making progress, the display jumping all over has stopped so I think I can get this working if I keep going :) just need to monitor a different player

Just to be clear, I'm not complaining about the dev(s) I'd be lazy in documenting too but this thread may help someone else.

dannixDanny, 2020-08-09:
Ok, got it working now. Will look into the widgets, sequences etc later.

Install the master code as per instructions to download, untar and execute install.sh. Run python configure.py and answer the questions about your setup so that it can create your config. Specifying a MAC here does not seem to take effect so must be specified in your docker run command. 

The docker run command needs changing so you can edit your layout pages etc from where you donloaded the code to, usually your home dir. I run this docker command, default command will just spin up the image that install.sh downloads including it's copy of layout pages and configs :

` /usr/bin/docker run **--rm** --network=host --privileged -v /var/log:/var/log:rw **-v /home/pi/pydPiper-master**:/app:rw dhrone/pydpiper:latest python /app/pydPiper.py --lms --driver ssd1306_i2c --width 128 --height 64 --i2caddress 0x3c --i2cport 1 --temperature celsius --lmsplayer 'b8:27:eb:40:10:be' --pages pages_ssd1306.py
`
-v /path/to/your/download/of/pydpiper:/app:rw so you can edit locally
--rm to auto remove the docker container after it has stopped, default is leaving containers, have a look with "sudo docker container ls -a" then do a "sudo docker container prune" to clean them up. Once --rm added you won't need to cleanup anymore.

Been a nice challange for me this, guess I've learnt some Docker too. I'm running a SSD1309 controller, works fine with the luma_i2c driver and ssd1306 page layout.

dhrone, 2020-08-13:
Looks like you've resolved your issue.  Thanks for sharing the steps with the community.  I moved to docker for some of the distros because I got tired of chasing all of the dependencies that broke when the distros got updated but it has created its own set of challenges!

I may move back away from it if I can work through how to better manage installation on Volumio, Moode, etc.

closed 2020-08-13 by dhrone

dannixDanny, 2020-08-13:
I don't think docker itself is an issue, I'd just update the install script to use `-v` and `--rm`, assume the current directory it's run from is the install directory to give a starting run command that works as intended. Update the install instructions to have the user run `sudo python configure.py` It's not stated, or if it is I missed it. 

It's clear this has evolved alot, it is a fantastic bit of software. There is no way I would have achieved what I wanted without it so I'm very grateful for your time.