dhrone/pydPiper #107: ssd 1306 display problem

yammh, 2020-09-16:
Hi dhrone,

I installed pydpiper in volumio and configured ssd1306, but the display does not work.
The process seems to  work normally. Please check the settings below and let me know if there is any further work to be done.

---------------------------------------------------------------------------------------------------
pydpiper.service settings

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw  dhrone/pydpiper:latest python /app/pydPiper.py --volumio --driver ssd1306_i2c  --width 128 --height 64 --i2caddress 0x27 --timezone 'Asia/Seoul' --temperature celcius --pages pages_ssd1306.py
---------------------------------------------------------------------------------------------------
---------------------------------------------------------------------------------------------------
volumio@volumio4:~/pydPiper-0.3-alpha$ ps -ef |grep pyd
root       831     2  0 05:24 ?        00:00:00 [kcopyd]
root     32462     1  3 07:31 ?        00:00:00 /usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw dhrone/pydpiper:latest python /app/pydPiper.py --volumio --driver ssd1306_i2c --width 128 --height 64 --i2caddress 0x27 --timezone Asia/Seoul --temperature celcius --pages pages_ssd1306.py
volumio  32565  2855  0 07:31 pts/0    00:00:00 grep pyd
---------------------------------------------------------------------------------------------------

[image]

dhrone, 2020-09-16:
Your I2C address seems to be incorrect according to your i2cdetect results.  That showed a single device on the i2c bus with an address of 0x3C.  You might try that address instead of 0x27.

If that doesn't work though, it would be good to verify that the display is wired correctly before trying to work with pydPiper.  Take a look at the [Luma.Oled Documentation](https://luma-oled.readthedocs.io/en/latest/intro.html).  Pay particular attention to the Hardware and Installation pages.  If you can get one of the simple example programs to work, it will be easier to get pydPiper working for you.

yammh, 2020-09-17:
> Your I2C address seems to be incorrect according to your i2cdetect results. That showed a single device on the i2c bus with an address of 0x3C. You might try that address instead of 0x27.
>
> If that [...]

Hi dhrone,

i changed address to 0x3c. And then, The display works well.
Thank you for your help.

closed 2020-09-17 by yammh