dhrone/pydPiper #108: 24 hour mode and Time update

suldobrasil70, 2020-10-01:
In order to display the time in 24 hour mode and update the time every minute, I have found good info on previous posts. I have edited the pydPiper.py and pages_lcd_16x2.py files. I did this on a fresh installation and BEFORE I ran the install.sh script. However no luck. Time still updates only every 5 minutes and time display is in 12 hour mode.
What do I do wrong or forget?
thanks in advance

dhrone, 2020-10-01:
It's hard to say without knowing what you did and which distro you are using.

Usually, when people have made changes to the content inside the pydPiper directory whether page file, config file, or other changes which has no effect, they are running the dockerized version and have not granted the docker container access to the underlying filesystem.

Here's an example...

```
/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py --volumio --driver hd44780_i2c  --width 80 --height 16 --i2caddress 0x27 --timezone 'US/Eastern' --temperature fahrenheit --pages pages_lcd_16x2.py
```

the important part is 

```-v /home/volumio/pydPiper:/app:rw```

This tells docker to mount /home/volumio/pydPiper into the docker containers /app directory which is where the code gets executed from.  If this flag is not on your command line or if the directory that you've placed your pydPiper instance is not correctly specified, the changes will not be accessible to the container.

dhrone, 2020-10-01:
You are missing the -v reference to the pydPiper files.

Try...

```
ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/pi/pydPiper-0.3-alpha:/app:rw dhrone/pydpiper:latest $python /app/pydPiper.py --lms --driver hd44780_i2c --width 80 --height 16 --i2caddress 0x27 --i2cport 4 $--timezone 'America/Sao_Paulo' --temperature fahrenheit --lmsplayer 'b8-27-eb-98-0f-c4' --pages pages_lcd_16x2.py
```

Note the addition of `-v /home/pi/pydPiper-0.3-alpha:/app:rw`.  This will mount your local version of pydPiper into the /app directory of the container.

FYI, I wasn't sure why you had a $ in front of python.  Do you have an environmental variable `python` that contains the path to your python interpreter?

suldobrasil70, 2020-10-03:
Works like charm. Thanks a lot for your kind support!
BTW the $ in front of python does not mean anything. When I had copied the text of the service file entry from the editor, it only let me copy what was on the screen, so I had to copy-paste in several sections and the $ added itself. No environmental variable.

closed 2020-10-03 by suldobrasil70