dhrone/pydPiper #109: Max2Play - Not showing track/radio info on synchronized device with Squeezelite Player

suldobrasil70, 2020-10-18:
Me again. Pydpiper runs well on my main Max2Play device. Squeezebox Server installed. It shows the track/radio info when playing, and the time only when stopped.
For multi-room purposes, I have set up another max2play device, which only runs Squeezelite as a player; this device is synchronized with the first (main) device, which has squeezebox server. The music selection occurs thru the web/phone interface with squeezebox server.
Here is the problem: The squeezelite-only device ONLY shows the time, as if nothing were playing!
In the pydpiper.service files of each device, I have written the MAC addresses of each individual device. All other settings are equal.
Any thoughts would be appreciated.

dhrone, 2020-10-19:
This will likely be related to the LMS metadata source not operating correctly.  I'll have to pull down the latest version of max2play and LMS to see if something on their side has changed.  FYI, this may be the same issue that is occurring with issue #60.

dhrone, 2020-10-20:
Can you confirm your current version of max2play and the version of Squeezebox server you are running?

suldobrasil70, 2020-10-20:
> 
> 
> Can you confirm your current version of max2play and the version of Squeezebox server you are running?

Max2Play Version 2.49
Raspberry PI 3
Logitech Media Server Version: 7.9.4 - 1601565762

dhrone, 2020-11-01:
Could you give me a summary of how you installed pydPiper for both of the max2play systems?  I'd also like copies of the pydpiper.service files and the pydPiper.cfg files?  You can drag them into the response to this message but will need to rename them to have a .txt extension.

suldobrasil70, 2020-11-02:
Thank you for taking the time to look into.
On both the LMS server as on the client max2play installations, I installed the same way, as per your instructions in the readme file:
sudo wget https://github.com/dhrone/pydPiper/archive/v0.3-alpha.tar.gz
sudo tar zxvf v0.3-alpha.tar.gz
cd pydPiper-0.3-alpha
- edited the pydpiper.py and pages..16x2.py python scripts to update the time every 60 s, and to show the time in 24 hr format.
- edited the pydpiper.service file. both are attached.
- both I2C ports are 4 and the I2C modules addresses are different one from the other (I tested the LCDs with other python scripts and they work)
-Last, I ran the install script:
sudo ./install.sh
- I did not find any pydPiper.cfg file on neither installation.

More comments:
- On the server installation, pydpiper starts up pretty quickly and the track/radio display works. On the client though, it takes almost 10 minutes for it to start and it only shows the time (as mentioned before)
- Since last week, the time is off by one hour. One hour ahead of actual time. In the service file you can see my timezone which is correct. The Pi's system times are correct though. Must have something to do with North America/Europe switching back from daylight saving last week. Brazil does not have any daylight saving any more.
Hope this helps
[pydpiper_service_server.txt](https://github.com/dhrone/pydPiper/files/5477593/pydpiper_service_server.txt)
[pydpiper_service_client.txt](https://github.com/dhrone/pydPiper/files/5477594/pydpiper_service_client.txt)

dhrone, 2020-11-02:
Ok, I see you are using the v0.3 alpha release.  See if you have a pydPiper_config.py file instead.  If yes, please send both client and server versions of this file.  

FYI, I'm looking to see what the LMS_Server variable is set to.  I suspect it is set to u"localhost" for both.  On the client you'll want it to be u"**ip address of server**".  Definitely an oversight of mine in the documentation.  Really the LMS_Server variable should have been settable from the command line in this version but I somehow left it out.

suldobrasil70, 2020-11-02:
See attached. As you assumed, in both files the LMS_Server variable is set to u"localhost".
So you're saying on the server I should leave as is, and on the client change to the IP address of the server? Which format? "192.168.13.197"?
BTW I had not been aware of any other release than the alpha 0.3. I just went straight by the book (readme file).
[pydPiper_config_py_client.txt](https://github.com/dhrone/pydPiper/files/5478003/pydPiper_config_py_client.txt)
[pydPiper_config_py_server.txt](https://github.com/dhrone/pydPiper/files/5478004/pydPiper_config_py_server.txt)

dhrone, 2020-11-02:
Yes, the line should look like this (on the client only).

`LMS_SERVER = u"192.168.13.197"`

The code has evolved a bit since I last issued a release.  You may find the latest main branch version more stable.  Admittedly, the documentation needs to be updated and a new issue released.  That said, I'm working on a Python 3.0 based version which likely will take priority over an interim release of the current pydPiper.

suldobrasil70, 2020-11-03:
With the LMS server's IP address in the pydPiper_config.py, it works well. Although it still takes several minutes for pydpiper to startup.
Python 3.0 version sounds like a plan.
Also, if you could update the documentation of the latest stable version, I'd appreciate.
Thanks for your help

closed 2020-11-03 by suldobrasil70