dhrone/pydPiper #110: VarErr

manolomalele, 2020-10-28:
Hi everyone, It's a little bit I wasn't here.
I write to ask (if it's possible) to help me to solve this issue.
HW: pi4
SW: latest MoodeAudio

As you can see on the images below (pages file and picture of the display), using on the Canvas the widget "codec", it shows "var err".
It should be displaying the filetipe (flac, mp3, dsd...) but it doesn't work.

Any suggestion? Using volumio, with te same custom pagefile everything is correct.

Thank you all in advance 

[image]

[image]

dhrone, 2020-10-29:
For Moode, pydPiper gets its data from the MPD daemon which does not expose a codec variable.

manolomalele, 2020-10-29:
Ok, Thank you. Obviously on volumio it is different, isn't it?

Any workaround to solve it? 

I also tried with 'filetype' but nothing...

dhrone, 2020-10-29:
Not an obvious one.  It's possible that the data you are looking for could be queried from MPD using the readcomments function but that is not a current feature of pydPiper.

manolomalele, 2020-10-29:
ok, thank you really much for the great job vou've done.
I'm ok with your answer, you can close this topic.

ty

closed 2020-10-29 by dhrone