dhrone/pydPiper #111: Can not Modify Display Setting + Rotate Screen

sanwaldx6x, 2020-11-04:
Hello I need help!

I try to modify the Pages Files, since a few days.
But everything I change, will not change in the Display.

I modify the Page Files in /home/volumio/pydPiper-0.3-alpha
I modify the Files pages_lcd_20x4.py and pages_ssd1306.py.
After I press CTL+X , and Restart with sudo systemctl restart pydPiper.
The Display is showing me again exact the same like bevore the modification.
Of course I did reboot also a lot of time.
I seems to me, that I use the wrong Page Files... because the File is modified but shows the old state.

The display is working. I can also chose between the pages files, when I Modify the "ExecStart..." ind pydpiper.service.
And restart with:

sudo systemctl stop pydpiper
sudo systemctl disable pydpiper
sudo cp pydpiper.service /etc/systemd/system
sudo systemctl enable pydpiper
sudo systemctl start pydpiper

I use Raspberry 2B
Volumio 2.834
Oled Display ssd1306
Hifiberry Amp2

I'm a beginner for Raspberry, so I hope i forget maybe a small thing like compiling or something.

I have also a second Question:
is it possible to rotate the display view 180 degree?

Daniel

dhrone, 2020-11-04:
It is likely that your docker run command does not expose the content of /home/volumio to the container.  Can you send your pydpiper.service file?  You'll need to rename the file to have a txt extention (e.g. pydpiper.service.txt) to attach it to this message.

sanwaldx6x, 2020-11-04:
Hi,

first of all, thank's for the very fast help!
I attached my pydpiper.service file.
I use v0.3-alpha. I have used the installion instructions form "https://github.com/dhrone/pydPiper"

By the way, is it possible to rotate the view of the Display 180°?

[pydpiper.service.txt](https://github.com/dhrone/pydPiper/files/5490083/pydpiper.service.txt)

sanwaldx6x, 2020-11-05:
Now, I got it to work, with your hint.

I used the Example from pydpiper.volumio.service from Master Data and modified it to:

ExecStart=/usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v **/home/volumio/pydPiper-0.3-alpha**:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py.

Maybe the wrong File is being installed. Or you could at a Description in the Manual, for Noobs like me. 
If it would be possible to rotate the display view 180°, then I would be complete satisfied.

dhrone, 2020-11-05:
You are in luck.  The driver you are using is from the `luma.oled` project which does support rotation.  pydPiper does not currently expose this capability but it is a simple code change to enable it.  You will need to edit pydPiper/displays/ssd1306_i2c.py line 47.  Change it from...

`		self.device = ssd1306(serial)`
to
`		self.device = ssd1306(serial, rotate=2)`

Be careful not to change the indent of the line.  It should line up with the line above it.

sanwaldx6x, 2020-11-05:
If've seen that it should be possible with the luma.oled, but I was not able to choose the right line.
It's working! 
Thank you very much!

closed 2020-11-05 by sanwaldx6x