dhrone/pydPiper #113: Build full documentation

evildmp, 2020-11-08:
If this is still and actively-maintained project, I'd be very glad to contribute to the project by getting a complete set of documentation at least started, using RST and Sphinx, to the standard of https://www.brachiograph.art.

I will need some help getting this running (with one of Pi Zero/2/3/4/ and Volumio/Rune/Moode/Max2Play and an SH1106 OLED display as a baseline example; I can work from there). So far, I am still working through the process.

dhrone, 2020-11-08:
Well, active is probably giving it too much credit.  The time I can spend on it varies widely based upon how busy I am with work and family activities.  That said, I would be grateful for any assistance you are willing to provide.  There is a decision to be made though in terms of where to invest your energies.

Over the last few months when I have had time to code, instead of spending time directly on the pydPiper code-based, I've been working on a substantial rearchitected version of pydPiper.  It is a ground up reimplementation in Python 3 that breaks pydPiper into multiple components to enable better maintainability.  It is also intended to expose functionality that is not music player focused but still useful to other projects such as the lcd/oled driver logic.

These components include:

| Project | Description | Status |
| --- | --- | --- |
| tinyDisplay        | A project that implements widgets and screen management | active development; **85% done**  |
| tinyDisplayDraw    | A graphical interface to design tinyDisplay screens | concept phase; no code completed  |
| [luma.lcd](https://github.com/rm-hull/luma.lcd)/ [luma.oled](https://github.com/rm-hull/luma.oled) | Moved all of the display related code into Richard Hull's excellent Luma.lcd and Luma.oled projects.  This means that any display supported by luma will work in the next version of pydPiper  | **100% done** |
| pyAttention        | Data services to listen for music metadata (Volumio, MPD, LMS, etc). Also supports other interfaces including REST and RSS | **60% done** |
| pydPiper V2        | Assembles all of the projects listed above into the next generation of pydPiper | concept phase; no code complete  |

While it might seem from this description that there is a ton of work to do in order to get to a release, most of the heavy lifting has been done and some of the work can be deferred.  For instance, tinyDisplayDraw is not really necessary as tinyDisplay already supports a Yaml format to describe screens.  My goal is to get to some kind of release before the end of the year.

So, the decision to be made is should we work to update the documentation for pydPiper or spend the energy on tinyDisplay and then the other associated projects?  Either way, I'm more than happy to help you get pydPiper working on which ever distro you need with the SH1106 display.  Let me know what your preference is and I'll try to support you.

evildmp, 2020-11-09:
This sounds even better and provides a cleaner path. Is there any code that is ready to be committed to https://github.com/dhrone/tinyDisplay?

dhrone, 2020-11-09:
Yes, it was not visible because the master branch was not the default.  I fixed that so you can see where the code stands.  The most mature module is tinyDisplay/render/widget.py

evildmp, 2020-11-09:
I now have a basic documentation kit set up at https://tinydisplay.readthedocs.io/en/docs/, based on [my docs branch](https://github.com/dhrone/tinyDisplay/compare/master...evildmp:docs?expand=1). That can be redirected to your repository at any time of course.

The next step for me is to get something working; once I have that I can work outwards from there.

Do you use IRC/Slack/something else that would be suitable for occasional synchronous collaboration?

dhrone, 2020-11-10:
Quick work!  Thanks.

It's going to be awhile before the system will be functional at the top level but you can take a look at the test routines to see what functionality already exists.  I'll see if I can spend some time on it this weekend.

As for Slack/IRC, I have never set something up to address that but can look into it when I have some spare cycles.

evildmp, 2020-11-10:
IRC's probably easier; http://freenode.net is the most popular IRC server and it takes a moment to set up a channel. In fact I have created _#pydisplay_ already. If you don't have an IRC client already, Freenode has a web interface: https://webchat.freenode.net (note that the channels is called _#pydisplay_, not _pydisplay_).

dhrone, 2020-11-11:
Ok.  I'm on freenode.  Also, I've added an email address to my github profile.  Feel free to reach me on it.

dhrone, 2020-11-23:
@evildmp I just wanted to provide a quick update.  I've spent some time getting a handle on the changes that need to be made in tinyDisplay to get it to a release but have not had sufficient time to implement them.  With the US Thanksgiving holiday this week, I plan to spend some additional time on the code so hopefully I'll be able to get a release out to you soon.

dhrone, 2020-12-08:
@evildmp. I'm almost code complete with tinyDisplay.  Latest code just committed to master.  

Items still on the todo list for 1st release:
  * Transition font dependencies to PIL ImageFont (most of the functionality in tinyDisplay/font.py has been superceded by the code I contributed to luma.core but I don't want another dependency if I can avoid it.
  * Implement scroll synchronization (coordinate between scroll objects)
  * Increase code coverage (currently 62%; goal > 95%)
  * Automate installation
  * Setup build pipeline
  * Readme
  * Clean up the in-code documentation

So, lots of progress made.  When you have time, take a look at the test scripts to get a feel for how the system operates.  I'm happy to find time to hop on IRC or zoom to go over the design and answer questions.

dhrone, 2020-12-22:
@evilmp, I'm going to move any further issue documentation over to the tinyDisplay repo.

closed 2020-12-22 by dhrone