dhrone/pydPiper #114: Query regarding WEH001602A support on Moode 6.7.1

gitmuir, 2020-12-16:
Hi @dhrone, I've been following your Moode installation instructions which remy1961 [posted](http://moodeaudio.org/forum/showthread.php?tid=461&highlight=pydPiper) to the Moode Audio forums.

I am making use of a **Winstar WEH001602A** OLED display in my Audiophonics "RaspDAC" (ES9038Q2M / Pi3B+) project, curiously the Audiophonics product listing ([here](https://www.audiophonics.fr/en/screens-vu-meters/winstar-weh001602a-ecran-oled-16x2-blanc-hd44780-p-8512.html)) states this is a **HD44780** display, however the spec sheet ([here](https://www.audiophonics.fr/images2/8512/13203765800176363412.pdf)) says that it is infact **WS0010**.

[image]

After a bit of confusion with the GPIO numbering (whoops), I'm confident I've got it wired correctly and used your Moode process to install with the following **pydPiper.cfg**

```
[STARTUP]
startup_msg_duration = 5

[DISPLAY]
display_driver = winstar_weg
display_pin_rs = 7
display_pin_e = 8
display_pin_d4 = 25
display_pin_d5 = 24
display_pin_d6 = 23
display_pin_d7 = 27
display_width = 80
display_height = 16
display_enable_duration = 0.1
pagefile = pages_weh_80x16.py
animation_smoothing = 0.1

[SYSTEM]
logfile = /var/log/pydPiper.log
loglevel = info
timezone = Europe/London
time24hour = false
temperature = celsius

[SOURCE]
source_type = moode
volumio_server = localhost
volumio_port = 3000
mpd_server = localhost
mpd_port = 6600
mpd_password = moodeaudio

```

This has resulted in the display illuminating, but I'm seeing some odd characters and 'static' noise on screen (video [here](https://drive.google.com/file/d/1adVEyAajOwyZ-qw3l9tH7_xPS8fOR7LD/view)).

At this stage I am starting to think...

1. This display isn't supported by the **winstar_weg** driver, or perhaps I should be using **HD44780**?
2. My earlier mistake at wiring has sent 5V to a place it shouldn't have gone...and fried something!

Very grateful for your thoughts, I think I'm close - but I would defiantly benefit from your advice. 

Thank you for continuing to develop this excellent project.

dhrone, 2020-12-16:
Yes, the WEH001602A does use the ws0010 controller and for pydPiper you should be using the winstar_weg driver.  The ws0010 controllers support hd44780 style commands but requires a different initialization sequence.

Your wiring appears correct and does match your configuration file.

I suspect your display has been damaged based on its behavior but it would be good to eliminate pydPiper as a potential source of the problem.  

Try getting a small test program working.  Here's a quick option for that.

Install luma.oled
Log into your Moode device and then enter the following command.
```
sudo -H pip3 install --upgrade luma.oled
```
Assuming luma.oled installs successfully we can try a short test program.

Start python3
```
pi@moode:~ $ python3
Python 3.7.3 (default, Dec 20 2019, 18:57:59) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
```

Then enter the following statements

```
from luma.oled.device import ws0010
from luma.core.interface.parallel import bitbang_6800
bb = bitbang_6800(RS=7, E=8, PINS=[25,24,23,27])
weh = ws0010(bb)
weh.text = 'Is it working?'
```
This should display 'Is it working?' on the top line of your display.  If that works, we can go back to debugging your pydPiper installation.  If it doesn't, let me know what behavior you see and we can try something else.

gitmuir, 2020-12-16:
Many thanks @dhrone, I've tried the above on the existing Moode installation (with **pydPiper** running) and nothing appeared on screen. 

I then tried again on a clean Moode install and when `weh = ws0010(bb)` is entered I see 2 forward slashes far right of screen 

`‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎/  /`

...then when `weh.text = 'Is it working?'` is entered, I see a flashing cursor icon! (no text)

`‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ _‎‏‏‎ ‎‏‏‎ ‎‏‏‎ ‎‏‏‎ `

🤪Weird. I guess I may need to invest in a new screen?!

dhrone, 2020-12-16:
You should not have pydPiper running during the test.  I'd recommend disabling it first, then start from a freshly powered on system.

FYI, when the `weh = ws0010(bb)` command finishes, the screen should be blank.

gitmuir, 2020-12-16:
Yes when repeated on the clean Moode install the result was the same; I've ordered a new screen and will report back once I'm able to test. 

Many thanks for your help.

dhrone, 2020-12-16:
Good luck.  I'm going to close this for now.  Feel free to reopen as needed.

closed 2020-12-16 by dhrone

gitmuir, 2021-01-06:
**It works!** Thank you so much for the troubleshooting assistance @dhrone; a faulty OLED all along.

Do you have any tips to clean up the underline / underscore that appears under everything here? Only appears on a track listing screen, eg. not the clock or stop / play.

[image]

Thanks again.

dhrone, 2021-01-06:
I assume that you are using pages_weh_80x16.py as your page file (or pages_weg_100x16).  That underline is a one pixel high song progress bar.  It probably shouldn't be used when playing a webradio source.  If you're comfortable making the edit, feel free.  Otherwise, let me know what file you are using and I'll provide an update.

gitmuir, 2021-04-22:
Sorry for the long delay in my response @dhrone, in pages_weh_80x16.py -- would I simply remove `('songprogress',0,15)` from the CANVASES section on the lines for **playartist_radio**, **playalbum_radio** and **playtitle_radio**?

dhrone, 2021-04-22:
Yes, you are correct.  I don't have the time to test if I made the edit correctly, but here is a file that has the suggested edit.
[pages_weh_80x16.py.txt](https://github.com/dhrone/pydPiper/files/6358294/pages_weh_80x16.py.txt)

You'll have to remove the txt extension if you are just copying the file over to your system.

gitmuir, 2021-04-22:
> Yes, you are correct. I don't have the time to test if I made the edit correctly, but here is a file that has the suggested edit. [...]

Worked a treat, thanks again!