dhrone/pydPiper #115: DEBUG:Could not connect to MPD

kosto-stavr, 2021-01-07:
Hi,
I am using pydpiper with RuneAudio and a 16x2 oled display configured to connect to MPD.
After my last system upgrade, the display opens normally (displays time), but have no connection with MPD.
At least this is what I understand from the log.
Any idea of what happens?

2021-01-07 12:25:07,109:INFO:pydPiper starting
2021-01-07 12:25:07,339:DEBUG:Loading display controller
2021-01-07 12:25:07,340:DEBUG:Loading music controller
2021-01-07 12:25:07,367:DEBUG:MPD musicdata service starting
2021-01-07 12:25:07,370:DEBUG:Connecting to MPD service on localhost:6600
2021-01-07 12:25:09,377:DEBUG:Music Controller Starting
2021-01-07 12:25:09,378:DEBUG:Loading pages_DigiOne.py as page file
2021-01-07 12:25:09,385:DEBUG:Initializing current system status update process.  System status will update every five minutes
2021-01-07 12:25:09,390:DEBUG:Weather service not enabled
2021-01-07 12:25:09,392:DEBUG:Weather service not enabled
2021-01-07 12:25:09,467:DEBUG:System status: Temp 38°c, Disk space remaining 49%, IP address 192.168.1.101
2021-01-07 12:25:09,706:DEBUG:Loading image splash
2021-01-07 12:25:09,711:DEBUG:Loading image progbar
2021-01-07 12:25:09,754:DEBUG:v out of range with value -1.  Should have been between 0 and 100
2021-01-07 12:25:17,423:DEBUG:Could not connect to MPD
2021-01-07 12:25:22,435:DEBUG:Connecting to MPD service on localhost:6600
2021-01-07 12:25:32,466:DEBUG:Could not connect to MPD
2021-01-07 12:25:37,473:DEBUG:Connecting to MPD service on localhost:6600
2021-01-07 12:25:47,503:DEBUG:Could not connect to MPD
2021-01-07 12:25:52,509:DEBUG:Connecting to MPD service on localhost:6600
2021-01-07 12:26:02,530:DEBUG:Could not connect to MPD
2021-01-07 12:26:07,537:DEBUG:Connecting to MPD service on localhost:6600
2021-01-07 12:26:17,559:DEBUG:Could not connect to MPD
2021-01-07 12:26:22,566:DEBUG:Connecting to MPD service on localhost:6600
2021-01-07 12:26:32,589:DEBUG:Could not connect to MPD
2021-01-07 12:26:37,596:DEBUG:Connecting to MPD service on localhost:6600
2021-01-07 12:26:47,621:DEBUG:Could not connect to MPD

(Thanks a lot for your software !)

dhrone, 2021-01-07:
It looks like MPD is not accepting connections on port 6600 which is the default for MPD.  Perhaps something has changed within Rune.  I'll have to take a look.  Could you confirm the version you are using?

kosto-stavr, 2021-01-07:
Thank you for your answer !
Actually RuneAudio is for a long time not updated.
One of the users kept the lights on with updates and enhancements.
Recently he repacked all of his achievements in a new version called rAudio.
Here is more info: [https://github.com/rern/rAudio-1](url)

PydPiper with the usage of MPD (no redis anymore) was working OK on his RuneAudio versions and also on his first image of rAudio.
On the latest image the connection error appears.
I am not very familiar with coding but does the different name matters ?

kosto-stavr, 2021-01-08:
OK
I found a working solution!
In mpd.conf I change the default “bind_to_address    "0.0.0.0"” to “bind_to_address        "127.0.0.1"” and in pydPiper.cfg I use also “mpd_server = 127.0.0.1”.

BTW
This log entry: 
_“2021-01-08 11:04:22,531:DEBUG:v out of range with value -1.  Should have been between 0 and 100”_ 
is something that I have to take care?

(Thanks again for your software !  I am looking forward to try your new tinyDisplay - if it works on my 16x2 oled display)