dhrone/pydPiper #119: PydPiper with Moode audio

kentlefley, 2021-02-09:
I've got PydPiper installed and running on Moode audio.  The large font displays such as play and stop and time are working as expected, however during play, all I get is a message saying "no active widget". Could you please point me in the right direction to solve this?  I expect that this is not an issue with pydpiper, but probably just a new user problem.
I have tried posting in the Moode forum a few times but I am getting no response.
Thank you in advance for your help.

dhrone, 2021-02-09:
It sounds like the MPD daemon may have been changed.  The "no active widget" message occurs when pydPiper can't find a sequence that is active in your page file given the current variables coming from the music service which for Moode is MPD.  What version or Moode are you running?

kentlefley, 2021-02-09:
I am currently running Moode 7.0 and the mpd version is 0.22.3_p0x3. 
Is there an older version of Moode the is known to work?

kentlefley, 2021-02-09:
The problem seems to be fixed.  Someone had a suggestion in the Moode forum to do the following:

Edit the pages file as follows:
In the sequence section, replace every "stream" by  "encoding", so that you have db['encoding'] instead of db['stream’]

Why does this work?

K

dhrone, 2021-02-10:
It's an issue related to the variability in how different music sources (Volumio, Moode, Rune, LMS) provide their data.  The pageFile you are using is referencing a variable ('stream') that does not get produced in the MPD driver.