dhrone/pydPiper #120: Support for performer/composer/oeuvre/part paradigm

PeterWurmsdobler, 2021-02-16:
Hello,

From looking through the code I can see that the musicdata retrieved from various players are following the standard ID3 tag paradigm of artist/album/song which may cover quite a lot of music but is not suitable for classical music. Could the musicdata database be extended to support different paradigms, or at least map:

performer -> NEW, e.g. "Alfred Brendel (1994), 
artist -> composer, e.g. Ludwig van Beethoven
album -> oeuvre (or work), e.g. "Piano Sonata No. 32 in C minor, Op. 111"
song -> part (or movement), e.g.  "II. Arietta: Adagio molto semplice e cantabile"

Of course, I do recognise that the source or data provider needs to support an additional tag such as performer. But if I am not mistaken, more recent versions of the ID3 tags recognise a recording_artist or performer.

Regards, 
peter.

dhrone, 2021-02-16:
There is no musicdata database per-se.  pydPiper is reliant upon the information source for the distribution you are using.  If it provides the variable, you can include it on a pageFile to render it to the screen.  

Examples of supported sources...
MPD
Volumio
Logitech Media Server

PeterWurmsdobler, 2021-02-16:
Thanks for the quick response. If pydpiper is tag agnostic that would be fantastic.
However, https://github.com/dhrone/pydPiper/blob/master/sources/musicdata.py seems to define a dictionary of tags which appears to be populated in the player specific derived classes. Hence, if a player supported tags such as `performer` or `composer`, adding a corresponding tag in the page may not be sufficient. The `musicdata` dictionary would need to be augmented, too?
peter

dhrone, 2021-02-16:
The current version attempted to create a normalized form of all of the data sources.  The new version is just going to pass it forward and let the pageFile deal with any variances.

PeterWurmsdobler, 2021-02-17:
Thanks for the update on your promising plans for the new version of the core machinery becoming tag agnostic and possibly deriving all tags needed from the definition of the `widgets`, `canvases` and `sequence` of the page files. The latter would be specific for a music player and device combinations; I could imagine that even the corresponding `sources` music player client could become more generic, too. I guess you still need to cater for permutations of display device (its resolution and interface used) and music players supported. If there is anything I can help with I would be happy to do so.

dhrone, 2021-02-17:
Possibly.  I should probably start an issue thread dedicated to the new version for people to hear about the plans and where they stand.  For now, details of how I have been thinking about the project are in issue #113.  FYI, the tinyDisplay project is now feature complete.   Current work is focused on pyAttention which has proven more conceptually difficult but I do have the MPD interface working.

PeterWurmsdobler, 2021-02-17:
Thanks, I'll have a look at #113 .