dhrone/pydPiper #122: 3 boxes right hand side on bottom row.

pkarza, 2021-03-21:
Well - after pulling my hair out a bit, I got this working...but the clock has no issues with the display, but the web radio gives three boxes on the second line.. is this normal?
[image]

dhrone, 2021-03-22:
Yes, it is expected.  That is the `songprogress` bar that shows how much of a song has been played.  When playing streams though it quickly fills up if the streaming source does not change the `elapsed` value.  If you want to get rid of that then you should remove `, ('songprogress',65,8)` from the `playartist_radio`, `playalbum_radio`, and `playtitle_radio` entries within the `CANVAS` section of your pageFile (presumably `pages_lcd_16x2.py`).

dhrone, 2021-03-27:
I think this issue is closed.  Feel free to reopen if you disagree.

closed 2021-03-27 by dhrone