dhrone/pydPiper #123: Scrolling speed

pkarza, 2021-03-27:
I read through issue scroll speed (13? I think) and understand that I need to change the speed ... but what exactly should I change it to in order to get a smooth readable scroll on hd44780?  Thanks!

dhrone, 2021-03-27:
I'm assuming you're referring to the effect section of a widget definition.

I'll use title as an example
`'title': { 'type':'text', 'format':'{0}', 'variables':['title'], 'font':'small','varwidth':True,'effect':('scroll','left',5,1,20,'onloop',3,65) }`

effect's value is a tuple composed of the following values.
type: The type of the scrolling effect ['scroll' or 'popup']
direction: The direction the scroll will move in ['left', 'right', 'up', 'down']
distance: The number of pixels to move each time scroll is updated
speed:  The number of calls to scroll before movement occurs
gap: How many pixels to leave between the beginning and end of content that is scrolling
hesitation type: When to pause the scrolling if desired ['onstart', 'onloop']
hesitation time: Number of seconds to pause scrolling when hesitate is occurring
threshold: How wide (or tall if scrolling up or down) the content of the widget needs to be before scrolling occurs.

If you wanted to slow the title down to half speed, you would change it to...
`'title': { 'type':'text', 'format':'{0}', 'variables':['title'], 'font':'small','varwidth':True,'effect':('scroll','left',5,2,20,'onloop',3,65) }`

If you wanted to run at 1/3rd speed...
`'title': { 'type':'text', 'format':'{0}', 'variables':['title'], 'font':'small','varwidth':True,'effect':('scroll','left',5,3,20,'onloop',3,65) }`

FYI, HD44780 LCD displays have very slow refresh times making pleasant scrolling effects difficult to achieve.  You may want to disable scrolling all together, or increase the distance and decrease the speed.  So, perhaps...
`'title': { 'type':'text', 'format':'{0}', 'variables':['title'], 'font':'small','varwidth':True,'effect':('scroll','left',20,5,20,'onloop',3,65) }`

pkarza, 2021-03-28:
Yes, I see what you mean about the slow refresh.  I did replace 'scroll' with popup and did nothing else... that didn't work... I don't know why - but I also used ('scroll','left',20,5,20,'onloop',3,65)  ... and that works.

Do I need to do anything else to make popup work?  I appreciate the support you have given.  Great addition to volumio.

dhrone, 2021-03-28:
Popup is is used to create an effect where the canvas is larger than the display and the popup effect displays the top part first and then animates the movement of the canvas as it moves up to display the bottom portion of the canvas.  This effect does not work well with HD44780s.  It does look good on the winstar WEH/WEG displays and any other small display with reasonable refresh rates AND full graphical support.

pkarza, 2021-03-29:
working great, thanks.

closed 2021-03-29 by dhrone