dhrone/pydPiper #125: Volumio 2.907 installation docker error

ddz169, 2021-09-07:
Installing Docker from the convenience script

# Executing docker install script, commit: 93d2499759296ac1f9c510605fef85052a2c32be

DEPRECATION WARNING
    This Linux distribution (raspbian jessie) reached end-of-life and is no longer supported by this script.
    No updates or security fixes will be released for this distribution, and users are recommended
    to upgrade to a currently maintained version of raspbian.

excelsiorHorizon, 2021-09-11:
Im having the same issue. Any help would be appreciated :)

dhrone, 2021-09-15:
Please let me know what specific distribution you are using.

excelsiorHorizon, 2021-09-15:
For me, I was able to get pydPiper working when I flashed my Pi with the latest build of Volumio Buster and did re-installed. 

The stable version Volumio 2.909 was giving me the docker issues.

Thank you so much for this contribution to the community :)

dhrone, 2021-10-14:
I do not currently have the time to resolve dependency issues with Volumio.  I understand that the new V3 version of Volumio may work but have not tested it.

Saiyato, 2022-02-08:
The problem with the other versions of Volumio is that Jessie is no longer supported by Docker, so installing Docker fails :wink: Not something @dhrone can fix I'm afraid. You'd either need to investigate using another runtime or find a ppa that has the old Docker binaries available.