dhrone/pydPiper #126: moodeaudio 7.4.1 pydpiper will not start

ulischlee, 2021-11-02:
I like to use pydpiper on moodeaudio for a longer time with a hd44780 20x4 display.
Now I made a fresh install with moode 7.4.1 and after the installation of pydpiper I test it with

sudo /usr/bin/docker run --network=host --privileged -ti -v /var/log:/var/log:rw -v /home/pi/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha /bin/bash
python pydPiper.py

it works fine.

But when I activate the pydpiper.service and reboot, pydpiper won't start.

The command systemctl status pydpiper shows:

moode systemd[ ]: Started pydPiper
moode docker[ ]: python: can't open file '/app/pydPiper.py': (Errno 2] No such file or directory

I think the docker container dies and gets restartet, till it dies again.

I've no idea what to do.

Regards Uli

dhrone, 2021-11-02:
Can you provide a copy of your pydpiper.service file?

ulischlee, 2021-11-03:
Hi,

the pydpiper.service file is attached.

It is unchanged from the convenience script. In former installations, I
tried to delay the start of docker or changed "after = volumio.service"
to another service, that I wanted to start earlier.

But nothing showed a positive effect.

I run moode on a Rpi3B with a 16GB sd card with an expanded file system.
In earlier installations I used a Rpi Zero. But it wasn't the lack of
power that caused the trouble with docker.

Regards

Uli

[Unit]
Description = pydPiper
Requires = docker.service
After = volumio.service

[Service]
Restart = always
ExecStart = /usr/bin/docker run --network=host --privileged -v /var/log:/var/log:rw -v /home/volumio/pydPiper:/app:rw dhrone/pydpiper:v0.31-alpha python /app/pydPiper.py

[Install]
WantedBy = multi-user.target