dhrone/pydPiper #130: Question - Is it possible to get this running on a Pi running picoreplayer only

TheHighFlyingBirds, 2022-01-31:
Hi,

Firstly, this is a fantastic program, even I managed to managed to get it running on a pi running volumio, and also managed to make some changes to get what I wanted until I killed something and it stopped working, so going to have to a fresh install.

Anyway, back to the question, I run a LMS server on one Pi, and then have picoreplayer on a second pi which has an official touch screen. I want to add a LCD screen to the pi running picoreplayer, to display system info and music data (bitrate etc). Is there any way of getting pydpiper running on picoreplayer? I recognise picoreplayer is a small OS and as such, does not support some programs, especially docker. If it is possible, is there any guidance anywhere?

Thanks

dhrone, 2022-01-31:
It should be possible.  The challenge is to either get Docker installed or alternatively to get all of pydPiper's dependencies installed.  

I don't have time at the moment but I'll add this as something to look into.

TheHighFlyingBirds, 2022-01-31:
Many thanks, would be brilliant if you could work your magic and get it working.

TheHighFlyingBirds, 2022-02-12:
Following on from above, is there a list of dependencies anywhere, as happy to have a play and see if I can get it working outside of docker.

TheHighFlyingBirds, 2022-02-21:
Hi dhrone,

I have been having a play with max2play and have managed to update the 'musicdata_lms.py' file so that samplerate and bitdepth (bitdepth only works on FLACs / files greater than 320kbps) can be displayed. I have included the lines below, as thought you could add it to the hosted file. The lines are added at line 284. Also added my modified file.

[musicdata_lms.py.txt](https://github.com/dhrone/pydPiper/files/8111608/musicdata_lms.py.txt)

try:
self.musicdata[u'samplerate'] = urllib.unquote(str(self.dataplayer.request("songinfo 2 1 url:"+url+" tags:T", True))).decode(u'utf-8').split(u"samplerate:", 1)[1]
except:
self.musicdata[u'samplerate'] = u""
try:
self.musicdata[u'bitdepth'] = urllib.unquote(str(self.dataplayer.request("songinfo 2 1 url:"+url+" tags:I", True))).decode(u'utf-8').split(u"samplesize:", 1)[1]
except:
self.musicdata[u'bitdepth'] = u""

I also commented out the bitdepth and samplerate lines under UNSUPPORTED VARIABLES.

This leads on to a quick question. The sample rate is coming out as a full number, e.g. 44100 (Hz), but I would prefer this to be in kHz. Is there a way to easily divide this number by 1000 to get the kHz value i.e. 44.1 (kHz)? I have tried various ways, but my lack of knowledge is not helping.

Thanks again.