dhrone/pydPiper #139: Volumio 3.695 has a socket.io update

SimonF-boop, 2024-07-24:
Hi

Volumio's recent 3.695 release included an update to socket.io v 2.3.0

[https://community.volumio.com/t/volumio-changelog/1446](url)

This seems to have broken the player state reporting. My current setup just shows a working clock, but nothing else is ever shown no matter what the player is doing. Rolling back to a version before the socket.io update and normal service is resumed.

from: sudo journalctl -f

```
Jul 24 18:19:00 volumio docker[27917]: pydPiper starting
Jul 24 18:19:00 volumio volumio[1034]: verbose: New Socket.io Connection to localhost:3000 from 127.0.0.1 UA: python-requests/2.18.4 Engine version: 3 Transport: polling Total Clients: 10
Jul 24 18:19:00 volumio docker[27917]: Exception in thread Thread-2:
Jul 24 18:19:00 volumio docker[27917]: Traceback (most recent call last):
Jul 24 18:19:00 volumio docker[27917]:   File "/usr/lib/python2.7/threading.py", line 552, in __bootstrap_inner
Jul 24 18:19:00 volumio docker[27917]:     self.run()
Jul 24 18:19:00 volumio docker[27917]:   File "/usr/lib/python2.7/threading.py", line 505, in run
Jul 24 18:19:00 volumio docker[27917]:     self.__target(*self.__args, **self.__kwargs)
Jul 24 18:19:00 volumio docker[27917]:   File "/app/sources/musicdata_volumio2.py", line 43, in run
Jul 24 18:19:00 volumio docker[27917]:     with SocketIO(self.server, self.port) as socketIO:
Jul 24 18:19:00 volumio docker[27917]:   File "/usr/local/lib/python2.7/dist-packages/socketIO_client/__init__.py", line 353, in __init__
Jul 24 18:19:00 volumio docker[27917]:     resource, hurry_interval_in_seconds, **kw)
Jul 24 18:19:00 volumio docker[27917]:   File "/usr/local/lib/python2.7/dist-packages/socketIO_client/__init__.py", line 54, in __init__
Jul 24 18:19:00 volumio docker[27917]:     self._transport
Jul 24 18:19:00 volumio docker[27917]:   File "/usr/local/lib/python2.7/dist-packages/socketIO_client/__init__.py", line 62, in _transport
Jul 24 18:19:00 volumio docker[27917]:     self._engineIO_session = self._get_engineIO_session()
Jul 24 18:19:00 volumio docker[27917]:   File "/usr/local/lib/python2.7/dist-packages/socketIO_client/__init__.py", line 76, in _get_engineIO_session
Jul 24 18:19:00 volumio docker[27917]:     transport.recv_packet())
Jul 24 18:19:00 volumio docker[27917]: StopIteration
```

Hope you can resolve this :)

Cheers

Simon

cedricp, 2025-02-07:
I updated the code for Python3 and changed some things to make it work with latest volumio3 version. It currently work with an HD44780 i2c.
I don't have others hardware, but I think it's not complicated to fix the remaining code.
https://github.com/cedricp/pydPiper
Hope that could help

SimonF-boop, 2025-02-07:
Any chance you can share how to configure? I've updated config.py in the pydpiper sub directory to the values reported by i2cdetect and run with "python3 -m app.pydPiper", but get an exception:

pydPiper starting
Loading font from /home/volumio/pydPiper/app/displays/fonts/latin1_5x8_fixed.fnt
Uncaught exception
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/volumio/pydPiper/app/pydPiper.py", line 721, in <module>
    lcd = displays.hd44780_i2c.hd44780_i2c(rows, cols, i2c_address, i2c_port, enable)
  File "/home/volumio/pydPiper/app/displays/hd44780_i2c.py", line 131, in __init__
    self.write4bits(0x33,False)
  File "/home/volumio/pydPiper/app/displays/hd44780_i2c.py", line 157, in write4bits
    self.bus.write_byte(self.i2c_addr, bits_high)
OSError: [Errno 5] Input/output error
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/volumio/pydPiper/app/pydPiper.py", line 721, in <module>
    lcd = displays.hd44780_i2c.hd44780_i2c(rows, cols, i2c_address, i2c_port, enable)
  File "/home/volumio/pydPiper/app/displays/hd44780_i2c.py", line 131, in __init__
    self.write4bits(0x33,False)
  File "/home/volumio/pydPiper/app/displays/hd44780_i2c.py", line 157, in write4bits
    self.bus.write_byte(self.i2c_addr, bits_high)
OSError: [Errno 5] Input/output error

Cheers and thx! :)

cedricp, 2025-02-07:
This is a good start, you need to check the address of the i2c display (mine is 0x3f), using i2c-detect for exeample)
The error you get is when the LCD is not found/connected

cedricp, 2025-02-07:
of course, set the good address in app/pydPiper.cgf config file

SimonF-boop, 2025-02-07:
Argghhhhh! sorry, i am an idiot! somehow changed the address to 0x28 instead of 0x27 (epic facepalm!)

Got the clock up on the test rig, so all good so far

cedricp, 2025-02-07:
I did some fixes sooner, please check you pulled them too

SimonF-boop, 2025-02-07:
Works like a charm on volumio 3.785 with a 20x4 44780 i2c LCD💯

cedricp, 2025-02-08:
Awesome !I will do some improvements soon

jazzboME, 2025-02-08:
@cedricp Also wanted to add that your version also works very well on a Winstar OLED 16x2 (after configuring some pins in `pydPiper.cfg`) with Volumio 3.785. Thank you for creating your fork!